ev3dev/ev3dev #358: Permissions issue with latest release

Svetomech, 2015-07-10:
I'm getting this error when trying to add permissions (following 'getting started' guide): usermod: group 'i2c' does not exist

dlech, 2015-07-10:
Try installing the `i2c-tools` package.

Svetomech, 2015-07-10:
Why is it not included?

dlech, 2015-07-10:
No one ever asked :wink:

Svetomech, 2015-07-11:
Btw, may be closed now. Though I am still a bit curious as to why I wasn't getting this error in previous versions.

ddemidov, 2015-07-11:
>  Though I am still a bit curious as to why I wasn't getting this error in previous versions.

The recommendation to add users to `i2c` group has only been added three days ago in ev3dev/ev3dev.github.io@a193678d6bf7c5297392dc058e7bdd91266bc41d.

dlech, 2015-07-11:
I've reverted that commit for now. I will add `i2c-tools` and the corresponding python i2c packages to the next image release and then add the `i2c` group back to getting started.

closed 2015-07-29 by dlech