ev3dev/ev3dev #359: CONFIG_SCHEDSTATS=y is needed for bootchart to work

ensonic, 2015-07-13:
Can we enable this with future kernel builds?

    /lib/systemd/systemd-bootchart --rel
    Failed to open /proc/schedstat: No such file or directory

Right now we only have

    systemd-analyze plot > file.svg

which does not offer a lot of detail :/

dlech, 2015-07-13:
The menuconfig help says: 

> If you say Y here, additional code will be inserted into the
scheduler and related routines to collect statistics about
scheduler behavior and provide them in /proc/schedstat.  These
stats may be useful for both tuning and debugging the scheduler
If you aren't debugging the scheduler or trying to tune a specific
application, you can say N to avoid the very slight overhead
this adds.

So, is the very slight overhead worth having this turned on all of the time for all users?

ensonic, 2015-07-14:
You are right, probably not since building the kernel is easier than I thought.

Omn the opic of bootcharting, unfortunately the systemd-bootchart crashes while writing the svg :/ Debugging ...

dlech, 2015-07-14:
> Just in case I get it working. Should I create a wiki-page somewhere documenting it...

Yes, please.

ensonic, 2015-07-21:
Just to link the bootchart woes: https://github.com/systemd/systemd/issues/583

dlech, 2015-09-07:
Should I mark this as "wontfix" and close?

ensonic, 2015-09-07:
Indeed makes sense. Feel free to attach a WONTFIX label.

closed 2015-09-07 by ensonic