ev3dev/ev3dev #360: I2C -Connection

kretzschauer, 2015-07-22:
I tried to connect an arduino uno via i2c  to the ev3 to use it as an multisensor for robotics. The arduino has 0x04 as address an is sending some Bytes to the i2c -Bus.
In the graphical IDE of the ev3 it works well but I'd like to write Python-Programms for the ev3.
It also works with topikachu-Library, but it seems very slowly. So I tried it with the ev3dev-lang-python version. But there was no connection, I tried:
>>> import ev3dev
>>> m=ev3dev.i2c_sensor("INPUT_3","0x04")
>>> m.connected
returns False

What is wrong?
Thanks!

ddemidov, 2015-07-22:
What happens when you do
```py
m = ev3dev.i2c_sensor("in3","0x04")
```
?  Or `ev3dev.i2c_sensor(INPUT_3,"0x04")` (no quotes around `INPUT_3`)?

dlech, 2015-07-22:
A custom I2C sensor won't work work with the ev3dev driver framework unless you write a driver for it. You can still use the standard linux I2C libraries for communicating with it though. 

See http://www.ev3dev.org/docs/sensors/using-i2c-sensors/ to help get you started.

kretzschauer, 2015-07-23:
Thank you for your quick response.
@ddemidov : no success 
So I used smbus like this
<pre><code>
#!/usr/bin/python
import smbus
import time
# I2C-Adresse des Arduino
address = 0x40
# Erzeugen einer I2C-Instanz und Oeffnen des Busses
# scheinbar PortNr +2 
ardu = smbus.SMBus(6)#Port 4

anz=200
# Lesen
while anz>0:
  daten=[]
  reg= 0x01
  anz-=1
  for i in range(8):
      d= ardu.read_byte_data(0x04,0x00+i)
      daten.append(d)
  print (daten)
  time.sleep(0.01)
</code></pre>
And on Arduino an array with dummy-data was send to the ev3:
<pre><code>
#include <Wire.h>
#define SLAVE_ADDRESS 0x04
//Dummy Data 
uint8_t RegisterMap[8];

void setup()
{
    Serial.begin(9600); // start serial for output
    Wire.begin(SLAVE_ADDRESS);
    Wire.onReceive(receiveData);
    Wire.onRequest(sendData);
    Serial.println("Ready!");
    for (int i=0;i<8;i++){
      RegisterMap[i]=10+i;
    }   
}
int val=0;
int flag=0;

void zeigeWerte()
{
  Serial.print(RegisterMap[1]);
  Serial.print("\t");
  Serial.print(RegisterMap[2]);
  Serial.print("\t");
  Serial.print(RegisterMap[3]);
  Serial.print("\t");
  Serial.print(RegisterMap[4]);
  Serial.print("\t");
  Serial.println(RegisterMap[5]);
  delay(300);
}
void updateWerte()
{
  if (RegisterMap[0]<250){
    for (int i=0;i<8;i++){
      RegisterMap[i]=RegisterMap[i]+10;
      }
    }
   else{
     for (int i=0;i<8;i++){
      RegisterMap[i]=10+i;
      }
    }
 }
void loop()
{
  updateWerte();
  //zeigeWerte();
  if(flag==1)
   {
     Serial.print("Empfang: ");
     Serial.println(val);
     flag=0;
   } 
}

void receiveData(int byteCount)
{
    while(Wire.available()>0)
    {
      val=Wire.read();
      flag=1;
    }
}
// callback for sending data
void sendData()
{
  Wire.write(RegisterMap[val]);
}
</pre></code>
So it seems working.

closed 2015-07-23 by kretzschauer