ev3dev/ev3dev #361: RuntimeError: no device connected:

ihsan-akin, 2015-07-23:
Hi.
I'm trying to run the ev3-print3rbot  with ./writer.py 

I'm getting error 

Traceback (most recent call last):
  File "./writer.py", line 484, in <module>
    main()
  File "./writer.py", line 475, in main
    wri = Writer(calibrate = True)
  File "./writer.py", line 77, in __init__
    self.calibrate()
  File "./writer.py", line 89, in calibrate
    self.mot_lift.rotate_forever(speed=-50, regulate='off')
  File "./writer.py", line 27, in rotate_forever
    self.stop_command = stop_command
RuntimeError: no device connected: Function not implemented

The information

```uname -rv```
3.16.7-ckt14-6-ev3dev-ev3 #1 PREEMPT Wed Jul 8 14:08:29 CDT 2015

image : ev3-ev3dev-jessie-2015-07-08.img

```systemctl status brickman.service -l```
â brickman.service - Brick Manager
   Loaded: loaded (/lib/systemd/system/brickman.service; enabled)
   Active: active (running) since Thu 2015-07-23 23:08:19 EEST; 19min ago
 Main PID: 334 (brickman)
   CGroup: /system.slice/brickman.service
           ââ334 /usr/sbin/brickman

Any idea?

Regards.

ddemidov, 2015-07-23:
`writer.py` [expects to find a motor at output port B](https://github.com/cavenel/ev3-print3rbot/blob/1fc634149c6e295e9ded7b1ee5d9d51211d8deca/writer.py#L71). Do you have it connected? In case you do, can you check that the following works (in python shell)?
```py
>>> from ev3dev import *
>>> m = motor(OUTPUT_B)
>>> m.connected
True
```

ihsan-akin, 2015-07-23:
Thanks a lot
I'd a mistake there. Corrected now.

closed 2015-07-23 by ihsan-akin