ev3dev/ev3dev #362: typelib for ev3devkit is missing on the images/packages

ensonic, 2015-07-24:
```
find /usr -name *.typelib*
/usr/lib/arm-linux-gnueabi/girepository-1.0/GIRepository-2.0.typelib
/usr/lib/arm-linux-gnueabi/girepository-1.0/GObject-2.0.typelib
/usr/lib/arm-linux-gnueabi/girepository-1.0/Gio-2.0.typelib
/usr/lib/arm-linux-gnueabi/girepository-1.0/GLib-2.0.typelib
/usr/lib/arm-linux-gnueabi/girepository-1.0/GModule-2.0.typelib
```

dlech, 2015-07-24:
Is it missing if it wasn't there in the first place? :laughing: 

Currently, ev3devKit is just statically linked to brickman and is not packaged on its own. So I'm changing the title of this issue to reflect what actually needs to be done.

dlech, 2015-07-24:
I should also mention that the typelib is created when you compile ev3devKit as a shared library.

> Build as shared library instead of static library: `-DBUILD_SHARED_LIBS=bool:Yes`

So to get the demo running now, use the `LD_LIBRARY_PATH` and `GI_TYPELIB_PATH` environment variables to point to the appropriate location.

ensonic, 2015-07-27:
Thanks, I'll try to do the debian package and send a PR once done.

dlech, 2015-07-27:
All of this comes from the way we are currently running programs from brickman, primarily in an effort to not run user programs as root.

[In brickman, we are calling...](https://github.com/ev3dev/brickman/blob/ev3dev-jessie/src/controller/FileBrowserController.vala#L73)

    openvt --verbose --switch --wait -- sudo --login --set-home --non-interactive --user=<owner-of-file> -- <program>

brickman runs on `tty1`, so `openvt` gets the next free tty (`tty2`) and switches to it. So this is probably what is changing the permissions. The `sudo` command is then run on this tty, but apparently, it is not changing ownership of the tty to the proper user.

I know that I was previously able to get graphical programs working this way, so something has changed. Not sure what might have changed though. It sounds like we need to open a new issue to improve the way that brickman runs programs.

In the mean time, the program(s) will work if you do this (from an ssh terminal):

    sudo chvt 2
    sudo conspy
    <login>
    ./<program>

ensonic, 2015-07-29:
What license can I use for ev3kit, LGPL?

dlech, 2015-07-29:
It is GPL2. Pretty much everything in ev3dev is.

dlech, 2015-07-29:
If you haven't created a package already, I can get it done fairly easily. Or if you would like to learn how we have been doing packaging for ev3dev, I'll let you do it.

ensonic, 2015-07-30:
I've started it already (see above), but I am stuck at this:
https://gist.github.com/ensonic/9eb425e66b108e9b7655

dlech, 2015-08-01:
I've just pushed out the ev3devKit packages to the ev3dev.org package archive. You will find the typelib by installing `gir1.2-ev3devkit-0.1`. 

There is a `ev3devkit-demo` package that includes some of the demos. I omitted the python demos because I haven't done any python packaging before. Although there probably isn't anything special about it for just and executable.

I probably need to add a -doc package as well if you think that might be useful.

closed 2015-08-01 by dlech