ev3dev/ev3dev #363: Bluetooth cannot be enabled

JaMiKl, 2015-07-26:
First of all: thanks for providing ev3dev, it's really great!

The only issue I have is that is not possible to enable bluetooth.

Setup:
ev3-ev3dev-jessie-2015-07-08.img
3.16.7-ckt14-6-ev3dev-ev3 #1 PREEMPT Wed Jul 8 14:08:29 CDT 2015

I have updated everything using the `atp-get` commands.

In the Bluetooth menu, instead of "Start Scan", is see the entry "???":

[image]

I started `bluetoothctl` and typed "list", but I do not get any controllers listed (I guess there should be one visible here?)

When I do not boot ev3dev but use the original Lego firmware, bluetooth is working - so it is obviously no hardware defect. 

Thanks in advance for your support!

dlech, 2015-07-26:
This happens sometimes on the first boot because things don't always start in the same/proper order. Have you tried again after rebooting the EV3?

JaMiKl, 2015-07-26:
Yes, I rebooted the EV3 several times, but still the same problem.

dlech, 2015-07-26:
Have a look at the output of `sudo journalctl` and see if there are any errors.

You should see something like this...

```
Jan 01 00:00:36 ev3dev systemd[1]: Starting LEGO MINDSTORMS EV3 built-in Bluetooth...
Jan 01 00:00:36 ev3dev hciattach[251]: Found a Texas Instruments' chip!
Jan 01 00:00:37 ev3dev hciattach[251]: Firmware file : /lib/firmware/ti-connectivity/TIInit_6.2.31.bts
Jan 01 00:00:37 ev3dev hciattach[251]: Loaded BTS script version 1
Jan 01 00:00:37 ev3dev hciattach[251]: texas: changing baud rate to 2000000, flow control to 1
```

and then later...

```
Jan 01 00:00:37 ev3dev kernel: Bluetooth: Core ver 2.19
Jan 01 00:00:37 ev3dev kernel: NET: Registered protocol family 31
Jan 01 00:00:37 ev3dev kernel: Bluetooth: HCI device and connection manager initialized
Jan 01 00:00:37 ev3dev kernel: Bluetooth: HCI socket layer initialized
Jan 01 00:00:37 ev3dev kernel: Bluetooth: L2CAP socket layer initialized
Jan 01 00:00:37 ev3dev kernel: Bluetooth: SCO socket layer initialized
Jan 01 00:00:37 ev3dev kernel: Bluetooth: HCI UART driver ver 2.2
Jan 01 00:00:37 ev3dev kernel: Bluetooth: HCI H4 protocol initialized
Jan 01 00:00:37 ev3dev kernel: Bluetooth: HCI BCSP protocol initialized
Jan 01 00:00:38 ev3dev kernel: Bluetooth: HCILL protocol initialized
```

botbench, 2015-07-26:
There are some EV3 bricks out there with new BT chips.  They have been
reported on the Lejos forums.  I have contacted LEGO several times to try
and get some sample hardware, but they are unable to give me a SKU.

What is the output of hciconfig -a?

Regards,
Xander

| I'd Rather Be Building Robots
| Current projects and ramblings: botbench.com
| Google+: gplus.to/XanderSoldaat

botbench, 2015-07-26:
Just for reference:

New BT modules will return this with hciconfig -a:
hci0: Type: BR/EDR Bus: UART
        BD Address: 00:16:53:4A:B7:8A ACL MTU: 1021:4 SCO MTU: 180:4
        HCI Version: 4.0 (0x6) Revision: 0x0
        LMP Version: 4.0 (0x6) Subversion: 0x1b55
        Manufacturer: Texas Instruments Inc. (13)

Old ones will return this:
hci0:   Type: BR/EDR  Bus: UART
        BD Address: 00:16:53:3D:07:F5  ACL MTU: 1021:4  SCO MTU: 180:4
        HCI Version: 2.1 (0x4)  Revision: 0x0
        LMP Version: 2.1 (0x4)  Subversion: 0x191f
        Manufacturer: Texas Instruments Inc. (13)

Source: Andy Shaw

If the BT chip is not seen at all by ev3dev, just use a Lejos image or the
1.07H/E firmware from LEGO and connect via WiFi over Telnet.

Regards,
Xander

| I'd Rather Be Building Robots
| Current projects and ramblings: botbench.com
| Google+: gplus.to/XanderSoldaat

dlech, 2015-07-26:
Interesting. So some EV3s have Bluetooth 4.0 now?

@JaMiKl, could you also please post the hardware version as reported by the LEGO firmware.

dlech, 2015-07-26:
Here is the relevant leJOS thread: <http://www.lejos.org/forum/viewtopic.php?f=18&t=7811&start=15>

Looks like we need to included the updated bluetooth firmware.

dlech, 2015-07-27:
I've added the new Bluetooth firmware file from the LEGO V1.07 firmware. Please upgrade with `sudo apt-get install ev3-firmware` and let us know if it works.

JaMiKl, 2015-07-27:
Thanks a lot for the quick answers! I will try it out directly when I'm back home next weekend!

JaMiKl, 2015-08-02:
Checking `hciconfig -a` shows me that I have the new BT chip.

I updated now using `apt-get update` and `apt-get upgrade` (did not work with `apt-get install ev3-firmware`)

It works now and I can pair with my laptop!

Thank you so much for the quick support!

closed 2015-08-02 by JaMiKl

dlech, 2015-08-02:
Great! I wrote `firmware-ev3` backwards, which is why it did not work, but `apt-get upgrade` upgraded the package anyway.