ev3dev/ev3dev #364: Lidar lite

boisfermer, 2015-07-26:
Hi,
I'm trying without much luck to get lidar lite ( http://pulsedlight3d.com/ ) to work with ev3. Lidar lite has an i2c interface. I've verified the device is working on a raspberry pi and the supplied test code produces the expected output. i2cdetect on the pi gives the following output.
 pi@rp1 ~/code $ sudo i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- 62 -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --

When I run i2cdetect on the ev3 the lidar lite address isn't recognized at all.
john@ev3dev:~$ ls /dev/i2c*
/dev/i2c-1  /dev/i2c-3  /dev/i2c-in1
john@ev3dev:~$ sudo i2cdetect -y 3
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- -- 

Has anybody got this device working with the ev3 or have any insight that might help?
Best, John.

dlech, 2015-07-26:
[Our friends as leJOS have got it to work with the EV3.](https://lejosnews.wordpress.com/2015/05/29/ev3-lidar-sensor/)

@gloomyandy says:

> The sensor is an i2c device. The tricky part is that it has built in 10K pull up resistors. This causes problems with the EV3 as the data line has a protection resistor in series with the pin. This in effect forms a potential divider with the pull up resistor when the EV3 tries to pull the data line down to zero volts and means that the sensor does not see a zero volt level. I worked around this by adding an extra pull down resistor of around 8K (I think), which enabled the logic zero to be detected but still allowed logic one to work. Ypu may need to experiment a bit to find a value that works for you. To get the device to work at high speed I reworked the leJOS i2c driver. I may write a further article about that at some point, but the weather is too good at the moment. I’m busy riding my bike rather than writing articles! If you want to discus this further please post on the leJOS forum as I think that is a better place for a discussion than here.

boisfermer, 2015-07-30:
Thanks, that worked. Works fine with i2c-tools.

I wouldn’t mind having a go at writing a driver for it, so it’s a bit more plug and play with ev3dev. I’m a total newbie when it comes to physical computing (ok with C++ though) but if someone could point me towards any docs on the subject I could give it my best shot.

Cheers,
John.

dlech, 2015-07-30:
Since this isn't a designed-for-LEGO-MINDSTORMS product, it won't work with the ev3dev I2C sensor drivers (i.e. **NXT/I2C** as described [here](http://www.ev3dev.org/docs/sensors/#i2c-sensors)). Rather it falls into the category of **OTHER/I2C**. 

I don't know of any existing linux driver classes that this fits into, so rather than writing a kernel driver, I think you would be better off just writing a library that uses the standard linux i2c library. Perhaps you could adapt one of the existing arduino libraries?

closed 2017-01-14 by dlech