ev3dev/ev3dev #365: launching apps from the file browser need to setup the permission for the new tty properly

ensonic, 2015-07-28:
From discussion in #362

All of this comes from the way we are currently running programs from brickman, primarily in an effort to not run user programs as root.

[In brickman, we are calling...](https://github.com/ev3dev/brickman/blob/ev3dev-jessie/src/controller/FileBrowserController.vala#L73)

```
openvt --verbose --switch --wait -- sudo --login --set-home --non-interactive --user=<owner-of-file> -- <program>
````
brickman runs on tty1, so openvt gets the next free tty (tty2) and switches to it. So this is probably what is changing the permissions. The sudo command is then run on this tty, but apparently, it is not changing ownership of the tty to the proper user.

ensonic, 2015-07-28:
I've found a workaround/fix and will send a PR for it shortly:

option 1)
```
openvt --verbose --switch --wait -- sh -c "tty >/tmp/mytty; tty=$(cat /tmp/mytty); chown -v user: $tty; sudo --login --set-home --non-interactive --user=user -- /home/user/ui_demo.sh"
```

It is a bit ugly since openvt does not report which vt it opened (it only prints it to stdout). It gets even uglier since we can't do ```tty=$(tty)``` as that would give us ```/dev/pts/0```. Using a file in tmp is not good either.

option 2)
```
openvt --verbose --switch --wait -- bash -c "read tty < <(tty); chown -v user: $tty; sudo --login --set-home --non-interactive --user=user -- /home/user/ui_demo.sh"
```

this uses bash since there we have the read command with the re-direction. Any preference?

dlech, 2015-07-28:
Option 2 is a little cleaner, but if you look at the brickman code, we need to get the tty back to brickman as well, so maybe the tmp file is not so bad. brickman can delete the tmp file as soon as it reads it. In that case, we can drop the `--verbose` option.

ensonic, 2015-07-28:
It seems to work (after many hours)
https://github.com/ensonic/brickman/commit/b025884128df52ac47d15eb8bbacd2b3fcd1a2b4

dlech, 2015-09-12:
This has been released in brickman 0.6.2

closed 2015-09-12 by dlech