ev3dev/ev3dev #366: brickman stdout/err to journal

ensonic, 2015-07-28:
Create this file.
```
# cat /etc/systemd/system/brickman.service.d/brickman.conf 
[Service]
Environment="G_MESSAGES_DEBUG=all"
StandardOutput=journal
StandardError=journal
```
Run ```systemctrl restart brickman.service``` and then ```journalctl -f -u brickman.service```. Now you'll see all glib warnings.

Ideally there are none, so we could consider doing this permanently. Any opinions.

dlech, 2015-07-28:
I only use `G_MESSAGES_DEBUG=all` when actually debugging brickman. I don't see a need to output so many messages. By default, glib outputs errors which is what I have been using mostly in brickman anyway.

I don't have a problem with piping the brickman output to journal.

ensonic, 2015-07-29:
Cool. I'll send a PR that adds the piping and I expand the README with the commands.

dlech, 2015-07-29:
Tip:

If you add `ev3dev/ev3dev#366` to your commit messages, GitHub will automatically link the commits to this issue. I usually make the last line of my commit messages `Issue ev3dev/ev3dev#NNN`. If you write "fixes" or "closes" before the issue, it will automatically close the issue when the commit is merged, but I don't like doing that on ev3dev. I would rather have the issue open until the fix has been released, not just committed.

ensonic, 2015-07-30:
Updated the ticket references to be "See issue ev3dev/ev3dev#xxx".

dlech, 2015-07-30:
Awesome!

dlech, 2015-09-12:
This has been released in brickman 0.6.2

closed 2015-09-12 by dlech

ensonic, 2015-09-24:
Thanks, confirming that it works.