ev3dev/ev3dev #370: [brickpi] Using set_device attribute to set ev3-uart sensor hangs if sensor is not connected

dlech, 2015-07-29:
Using brickpi on RPi with nothing plugged into the sensor ports. Steps to repeat....

    cd /sys/class/lego-port/port0 # presuming this is a brickpi input port
    echo ev3-uart > mode
    echo lego-ev3-us > set_device # this command hangs

The command hangs until a sensor is actually plugged in.

dlech, 2016-02-04:
Cannot duplicate anymore. There is a long timeout (10 seconds) but does not hang. May have been fixed by commit from #531.

closed 2016-02-04 by dlech

dlech, 2016-02-04:
Now I'm getting a timeout of nearly a minute. This is due to the BrickPi just not responding at all. It's like the microprocessor on the BrickPi is in a bad state. This will probably have to be fixed in the BrickPi firmware rather than in ev3dev.

jabrena, 2016-02-04:
Good morning,

What is the process to update the firmware on brickpi?

dlech, 2016-02-04:
http://www.dexterindustries.com/BrickPi/about/design/brickpi-firmware-update/

Although if you want better firmware, you have to change the firmware source code yourself. :wink: