ev3dev/ev3dev #373: gi bindings for grx are needed

ensonic, 2015-07-30:
One can use ev3devkit just fine from e.g. python, but one could not write apps like brickman. E.g. one cannot load fonts. Since we have a copy of grx in our repo anyway, we could add basic gboxed types and add gobject introspection for it. Two example libraries that do these are harfbuzz and graphene:
https://github.com/behdad/harfbuzz/blob/master/src/hb-gobject-structs.cc
https://github.com/ebassi/graphene/blob/master/src/graphene-gobject.c

I'll give this a try.

dlech, 2015-07-30:
FYI, GRX is abandoned since the end of 2012, so you can change as much as you want.

dlech, 2015-10-19:
Another thing to consider is that GLib uses UTF8 strings everywhere. After fixing the recent performance issues, I noticed the next biggest CPU eater were the `vala_string_width` functions from the GRX vapi. The underlying `string_width` in GRX doesn't actually take that much time, but the generated code from vala does.

There is a fork of GRX called [MGRX](http://www.fgrim.com/mgrx/) that supposedly has UTF8 support. This would be good for internationalization. There are only a couple of bit-mapped Unicode fonts available (not counting fixed-width console fonts). There is [unifont](http://unifoundry.com/unifont.html) (debian package `xfonts-unifont`). However it is only available in a 16 pixel height, which is a bit big for our small screen size. That's only 8 rows of text with approx 20 chars per line. There is also [efont](http://openlab.ring.gr.jp/efont/unicode/) (debian package `xfonts-efont-unicode`) that does not have quite as much coverage, but has a wider variety of sizes. I don't think GRX or MGRX has support for reading fonts from .pcf file (provided by the `xfont-*` packages), so support for that would have to be added.

dlech, 2015-12-03:
I've made quite a bit of progress on this, but there is quite a bit more to make it usable. Everything I have done so far is in the [grx3 branch](https://github.com/ev3dev/grx/commits/grx3).

For the types that need it (context, pattern, image, font), I have been make into GLib boxed types as suggested.

Things that need to be done to make this actually usable via introspection:

- [ ] Add documentation comments with annotations for all public functions. Documentation should be copied from <http://grx.gnu.de/grx249um.html> as much as possible when we do this.
- [ ] Write a demo program in python that uses gobject introspection to test this out and make sure everything works.

Things I would still like to do:

- [ ] Remove the existing font stuff and use pango instead. This seems like the easiest way to get UTF8 support.
- [ ] Make functions more GLiby. 
    - [ ] Create functions should be constructors, but constructors can't return null, so this has to be changed.
    - [ ] Use GError for passing errors (throwing exceptions)
    - [ ] Make return values consistent. Some functions return true/false, others 0/-1 for success/failure.
- [ ] Fixup the input framework to use evdev. I would also like to consider merging in code from [tslib](https://github.com/kergoth/tslib) as part of the input framework.

Any help here would be appreciated.

dlech, 2015-12-03:
Links for documentation comments and annotations:

* https://developer.gnome.org/gtk-doc-manual/unstable/documenting.html.en
* https://wiki.gnome.org/Projects/GObjectIntrospection/Annotations/Old

dlech, 2017-09-01:
I think we can close this now. :smile: 

https://www.youtube.com/watch?v=mQE_rWU5E18

GRX3 has been officially release.

closed 2017-09-01 by dlech

jabrena, 2017-09-01:
Nice development!!!