ev3dev/ev3dev #374: nxt-touch sensor problem on BrickPi

Ho33e5, 2015-07-31:
I have a raspberry pi 1 B (rev2) with [this](https://github.com/ev3dev/ev3dev/blob/master/release-notes/rpi-ev3dev-jessie-2015-05-13.img-release-notes.md) ev2dev image.

What I do to manually load the sensor:
```bash
$ echo nxt-analog > /sys/class/lego-port/port0/mode
$ echo lego-nxt-touch > /sys/class/lego-port/port0/set_device
```

The device `/sys/class/lego-sensor/sensor1` (why not `sensor0`?) appears, but the `value0` property doesn't reflect the state of the touch sensor (it's blocked to `1`).

Interestingly, the `bin_data` property seem to work a bit more: when the button is pushed i get a random byte between `\xd8` and `\xdc` for the first byte and `\x02` for the second. When the button is released i get the first byte between `m` and `r` and `\0x03` for the second. Is this the expected behavior?

dlech, 2015-07-31:
> why not sensor0?

When you set the port to `nxt-analog` mode, it automatically loads a generic driver (also called `nxt-analog`), so this is `sensor0`. When you use `set_device`, the `nxt-analog` sensor is removed and a new sensor is created, so you get `sensor1`.

I'm not seeing any change at all, so there is definitely something wrong here. At least you are seeing the values in `bin_data` changing, which is a good sign. Tip: you can use `od` to get hex values from `bin_data`.

I'll look into this a bit more later.

Ho33e5, 2015-08-02:
Thanks for the clarification on the `sensor0` thing.

laurensvalk, 2015-08-04:
There exist two types of NXT Touch Sensors. The older type does not support the auto-detect feature of the EV3. Possibly this is why the Brick Pi cannot see it currently?

If I recall correctly, in the newer type both the red and black wired are internally connected within the sensor, such that both are ground, which helps the EV3 identify it as this type of sensor. The older sensor did not have this.

laurensvalk, 2015-08-07:
I tested some of my NXT touch sensors to verify the above:

- This also effects the EV3 brick, not just brickpi. When using the older NXT Touch Sensor, the brickman status is "no-sensor" when the sensor is not pressed. It is "error" when the sensor is pressed.

- Testing with a multimeter, it looks like the old Touch Sensor is just a switch across pins 1 (white) and 3 (red). Pressing the sensor closes the switch. (R = 2.15 kOhm). Releasing the sensor opens the switch (R = inf). Pins 2 (black) and 3 (red) are not connected (R = inf).

- The newer Touch Sensor has an internal connection between pin 2 and 3 (R = 0). The switch is still across pins 1 and 2/3 (When pressed, R = 2.15 kOhm).

I suppose that rules out auto-detect for the old Touch Sensor, but it can be configured as a simple switch. (I haven't checked, but I'm guessing there's already a sensor driver for that?)

dlech, 2015-08-09:
@laurensvalk, this is good info (I've added a note to the sensors page in this regard), but the BrickPi has no autodetection at all, so it cannot detect *any* type of sensor.

antonvh, 2016-01-15:
Could it be a problem in the address naming in the driver? I find it weird that the address of a motor starts with `ttyAMA0` but the address of an analog sensor on the brickpi doesn't. It's just `in1`, not `ttyAMA0:in1` as I'd expect.

antonvh, 2016-01-15:
This is what I did, for the record:
```
$ python
>>> p = Device('lego-port', name_pattern='*', address='in4') 
>>> p.connected
True
>>> p.set_attr_string('mode', 'ev3-analog')
>>> p.set_attr_string('set_device', 'lego-ev3-touch')
>>> s = Sensor(address='in4') #Now that the lego port is initialised a lego-sensor appears in /sys/class
>>> s.connected
True
>>> s.get_attr_int('value0')
0
```

The last result is always zero, whether I press the sensor or not.

antonvh, 2016-02-09:
I ran into this again with another sensor. How can I test the fix? Should I do `apt-get dist-upgrade`? Or wait for a new image?

This is what I did:

```
$ cd /sys/class/lego-port/port5/
$ echo nxt-analog > mode
$ echo ht-nxt-gyro > set_device
$ cd in4\:ht-nxt-gyro//lego-sensor/sensor1
$ while true; do cat value0; done
```
result: -540

dlech, 2016-02-09:
I haven't officially published a kernel with the fix yet, but you can find a package for testing [here](https://github.com/ev3dev/ev3dev/issues/531#issuecomment-180933735).

dlech, 2016-02-09:
And yes, this issue affects all NXT/Analog sensors, not just the LEGO NXT Touch sensor.

dlech, 2016-04-11:
This fix has been released in the `4.1.21-10-ev3dev-rpi` kernel. Be aware the tacho-motor class has breaking changes in this release, so don't expect motors to work with your existing programs/libraries if you upgrade to this version.

closed 2016-04-11 by dlech