ev3dev/ev3dev #375: Bandwidth on TP-LINK TL-WN725N very limited

G33kDude, 2015-07-31:
I just ordered and received a TL-WN725N wifi adapter from TP-Link, intending to see how well it worked on my EV3.

I am currently downloading a Xubuntu ISO as a test. On my laptop it's downloading at a nice cozy ~1200KB/s. On the EV3, however, it's downloading at ~60KB/s.

G33kDude, 2015-08-01:
I went back to `ev3dev-jessie-2015-02-24.img` and I'm not sure if it's any better. I get ~60KB/s on the USB hub where I was testing it before, and about ~150KB/s plugged directly into the brick. I didn't test directly on the brick with whatever image I had before, but I can flash the newer firmware and test again to make sure I know what version it was I was testing to start with.

I went and found my Edimax dongle (which is held together by aluminum foil since I snapped part of the antenna) for comparison and it's getting ~650KB/s both on the hub and on the brick.

I might just buy a few more Edimax dongles, since they seem solid other than their brittle plastic that breaks off in a few months. It happened on the ones I bought, as well as seemingly everyone else's.

rhempel, 2015-08-03:
I have the 725 as well - worked great before and now it's not so good. I'll look at older drivers when I get back after August 11.

dlech, 2015-09-07:
Fix has been released in kernel `v3.16.7-ckt16-7-ev3dev-ev3`.

closed 2015-09-07 by dlech