ev3dev/ev3dev #376: No drivers for ‘Remote NDIS Compatible Device’ under Windows 10

GeQi, 2015-08-01:
I've been following the steps from the getting started section in the docs up to the part of [setting up internet connection](http://www.ev3dev.org/docs/getting-started/#step-4-setup-a-network-connection)

In the step No.8 there are suppose to be a driver for ‘Remote NDIS Compatible Device’ provided by 'Microsoft Corporation':
[image]

However under Windows 10 neither 'Microsoft Corporation' shows up among the Manufacturers nor the driver could be found anywhere

I know nothing about EV3 and Embedded System, am I doing something wrong or the guide just needs an update?

WasabiFan, 2015-08-01:
It looks like they changed this since earlier Windows 10 builds... at one point they had removed "Microsoft Corporation" and put it all under "Microsoft", but it doesn't look like the one we want is there anymore. I can't find any docs online about it, so we might have to ask a question on Microsoft's forums to see if anyone knows -- this is a pretty uncommon thing to want to do. @dlech do you have any ideas as to where it might be?

dlech, 2015-08-01:
It looks like it is broken for now. It looks like the driver is there, but I can't figure out how to install it.

Someone has already asked the same question on [MS Answers](http://answers.microsoft.com/en-us/windows/forum/windows_10-networking/windows-10-vs-remote-ndis-ethernet-usbgadget-not/cb30520a-753c-4219-b908-ad3d45590447). I'm subscribed to the thread, so hopefully someone will come up with a solution soon.

WasabiFan, 2015-08-01:
@dlech before you select the option to look through a list of drivers, I think there is a box where you can give it the path to the driver that you want to use. I don't have access to my PC at the moment, but I suspect that we could force it to install from that.

A few months back, I spent some time trying to install the drivers from PowerShell -- the only thing I found was some Windiws APIs that could be invoked. I suspect that that will be more trouble than it's worth.

G33kDude, 2015-08-01:
We could put the driver file in the windows readable partition of the sd card image. I'm still not clear on why windows doesn't pick up on the right driver automatically though

WasabiFan, 2015-08-01:
> I'm still not clear on why windows doesn't pick up on the right driver automatically though

This has been a problem in Windows 7 and 8 as well. The issue is that Acer has published a driver that claims to be compatible with RNDIS devices, and it automatically gets registered, even though it doesn't do what we need.

> We could put the driver file in the windows readable partition of the sd card image.

Putting a Windows system driver on an SD card and loading it on other machines is a _very_ dangerous idea. If you load the Windows 7 driver in Windows 10, you are voiding your Windows license and de-stabilizing your PC -- having people do that as part of a getting started guide isn't viable from my perspective.

G33kDude, 2015-08-02:
Voiding how exactly? Loading drivers from external sources is a pretty common occurrence pretty much everywhere, isn't it? It doesn't necessarily have to be the official Microsoft one, just something compatible.

WasabiFan, 2015-08-02:
> Voiding how exactly?

It voids volume licensing and associated warranties -- mostly for corporate-owned Windows installations, though.

> Loading drivers from external sources is a pretty common occurrence pretty much everywhere, isn't it?

I am not familiar with any normal cases where one installs a driver that didn't come from either Windows Update or a manufacturer's disk, and when it comes on a disk it's an installer and not the pure `.sys` and `.inf` files.

My worry is that there are a lot of cases where taking official Windows driver files from one PC and manually installing them on another can leave you in a bad state:
- If the version that you copy has had a Windows Update applied to it that the other files on the destination PC don't have
- If the normal driver installer does some extra config that you don't do yourself (e.g. add registry keys)
- If the source driver is from a different version of Windows
- If the source driver isn't signed with a certificate that the destination can verify (in that case you need to boot into safe mode with the driver verifier disabled).

> It doesn't necessarily have to be the official Microsoft one, just something compatible.

That's a good point; If we could find a third-party driver that had a publically-available installer, that should be fine. But we'd need to make sure that it is from a trusted vendor and is actively maintained, to make sure that there aren't any incompatibilities. And generally, I'm not sure that I would be willing to install a driver that I got like this, so I think we'd still need a separate method for people that didn't trust our version.

G33kDude, 2015-08-02:
> I am not familiar with any normal cases where one installs a driver that didn't come from either Windows Update or a manufacturer's disk, and when it comes on a disk it's an installer and not the pure .sys and .inf files.

I downloaded a printer driver installer a while back, and chose to extract the pure .inf and friends, putting them up on the company file share. I then installed a good 20-30 company desktops with them, and nothing seems to have gone wrong. Should I be concerned?

I would think/hope that instead of accepting and then self-invalidating that windows would just reject the invalid files. I can't imagine doing anything else from a customer support standpoint.

WasabiFan, 2015-08-02:
> I downloaded a printer driver installer a while back, and chose to extract the pure .inf and friends, putting them up on the company file share. I then installed a good 20-30 company desktops with them, and nothing seems to have gone wrong. Should I be concerned?

If you downloaded their installer and it had an option to dump the base files, that's a case that they have tested and they know that it works (as I mentioned earlier on the topic of finding a third-party driver). I'm concerned with copying Windows drivers to other PCs, where there is no standalone installer and it is updated through Windows Update along with the rest of the system.

> I would think/hope that instead of accepting and then self-invalidating that windows would just reject the invalid files. I can't imagine doing anything else from a customer support standpoint.

Windows has no way of knowing if they are invalid. See the warning that it gives you when installing a new driver (like we instruct people to do in our current tutorials):
[image]

They might be able to confirm that the driver is for the right architecture, but they can't make sure that it doesn't depend on other files or registry entries.

dlech, 2015-09-10:
I finally found a workaround for this. It is any *Easy 20 Step Process* TM. 

http://www.ev3dev.org/docs/tutorials/connecting-to-the-internet-via-usb/?os=Windows&os-version=10

If there are any errors in the instructions, please open a new issue - or even better, a pull request to fix them.

closed 2015-09-10 by dlech

capabilitysupportservices, 2024-06-24:
Capability Support Services, a leading [NDIS Provider Gold Coast](https://capabilityss.com.au/ndis-provider-gold-coast/), offers personalized and compassionate care to help you achieve your goals. Our dedicated team provides a range of support services, ensuring you receive the best care tailored to your needs. Contact us today for expert NDIS support.

adcservice, 2024-07-19:
ADCS offers compassionate [STA respite care in Adelaide](https://adcs.au/ndis-sta-respite-care-adelaide), ensuring a supportive and comfortable environment for all our clients. As a trusted NDIS provider in Sydney, we pride ourselves on delivering top-notch services, including NDIS Community Participation and NDIS Respite Care. Our dedicated team is committed to enhancing the quality of life for individuals and their families, offering personalised care and support tailored to each client's unique needs. With ADCS, you can be confident that your loved ones are in safe hands, receiving the highest standard of care and attention. Experience the difference with our professional and caring approach.