ev3dev/ev3dev #377: It is difficult to draw text on the LCD. Will there's a method to use "drawString" to draw string like LeJOS?

mob41, 2015-08-12:
I recently tried the ev3dev on my LEGO Mindstorms EV3. I found that it is difficult to program the EV3's LCD by writing the bytes in each pixels. And I saw the Wiki in LeJOS about LCD. There's a method to drawString into the LCD using Java. Is it possible to add this method for the future?

ddemidov, 2015-08-12:
If you use python bindings, either [official](https://github.com/ddemidov/ev3dev-lang-python) or [topikachu's](https://github.com/topikachu/python-ev3), then those let you draw on the LCD using python image library functions. Look at [these](https://github.com/ddemidov/ev3dev-lang-python/blob/master/demo/lcd.py) [examples](https://github.com/ddemidov/ev3dev-lang-python/blob/master/demo/motor-control.py#L39).

Otherwise, you could use your library of choice to create a bitmap image and then copy it to the framebuffer.

mob41, 2015-08-12:
Thank you. Will you add Java bindings?

ddemidov, 2015-08-12:
I won't, but may be you or someone else will :).

mob41, 2015-08-12:
It is impossible for me to program for java bindings because I am not a professional programmer :D

dlech, 2015-08-12:
leJOS uses the same display driver as ev3dev, so maybe you could use LCD code from leJOS on ev3dev.

Or you can use the display in console mode and just write text to standard out and it will display on the screen.

WasabiFan, 2015-08-12:
@mob41 To clarify, the ability to draw text on the display is not functionality that the kernel, in this case ev3dev, is responsible for. The kernel, along with its LCD driver, provide the support for raw access to the display so that higher-level applications and libraries can write custom code to display content.

What @ddemidov and @dlech are saying is that your ability to draw text on the display depends on the libraries that you use, and the kernel will support pretty much all Linux display libraries that can write to a framebuffer. So, in your case, you will want to find a library for the language that you are using that includes the ability to draw primitives to the display (basic shapes, text, etc.). 

If you are looking to write in Java, as far as I know we don't have any pre-made libraries for that (although given that LeJOS uses Java, I expect that supporting Java would be helpful for people coming from other platforms). You can always roll your own set of helper functions that read from and write to the necessary sysfs files (see [here](http://www.ev3dev.org/docs/tutorials/tacho-motors/)), which won't take much programming expertise as long as you just implement the functionality that you need in your app.

Alternatively, you could try out a different language that has similar syntax; C++ and JavaScript are both based on C syntax (which Java is also based off of), so you probably wouldn't have much trouble getting started with that. Python is also pretty easy to learn if you are looking to expand your programming skills :wink:.

closed 2016-04-07 by mob41

theZiz, 2016-04-07:
Fyi: The C lib https://github.com/theZiz/ev3c can draw lines, circles and text, too. I thought maybe it is the easiest way to write a simple Java wrapper for a C lib. ;)

mob41, 2016-04-08:
== This issue is one year ago.