ev3dev/ev3dev #378: How to use the ev3dev python bindings in eclipse PyDev?

mob41, 2015-08-12:
How to import ev3dev python bindings libraries' docs to eclipse PyDev?

ddemidov, 2015-08-12:
First, this belongs to [ddemidov/ev3dev-lang-python](https://github.com/ddemidov/ev3dev-lang-python) issues.

Second, I have no experience with eclipse IDE, but I suspect what you need is install python-ev3dev on your host computer. In case its a linux machine, see installation instructions in the [README](https://github.com/ddemidov/ev3dev-lang-python#installation). 

I also suggest you have a look at the recent IRC conversation on this topic: http://ev3dev.org/irclog/#id-23914

ddemidov, 2015-08-12:
Here is a copy of the relevant part of the IRC log: https://gist.github.com/ddemidov/1beb13d574b2f1c37507

WasabiFan, 2015-08-12:
It looks like this is what you're after (which seems to back up @ddemidov's thoughts): <http://stackoverflow.com/questions/3675585/how-to-add-python-libraries-to-eclypse-and-pydev>

mob41, 2016-04-27:
I just checked it out that why this issue is still open. almost a year ago...

closed 2016-04-27 by mob41