ev3dev/ev3dev #379: dFlex sensor support ?

GJpy, 2015-08-15:
Hi All,

I started to work with ev3dev few days ago and this is really cool ! I'm playing around with all kind of sensors. For my next project I'd need a dFlex sensor. It seems that it's not supported by ev3dev yet ? Any plan to have driver for them in the futur ?

the sensor:
http://www.generationrobots.com/en/401182-dflex-flexible-sensor-for-lego-mindstorms-nxt-robots.html#description

Thanks !

dlech, 2015-08-15:
According to <http://www.dexterindustries.com/manual/dflex/>, this is an NXT/Analog sensor. What happens when you plug it in? I am guessing that it should show up as a generic analog sensor (`nxt-analog` driver). It will just give you the raw value in mV, so it is up to you to scale it as needed.

GJpy, 2015-08-15:
Thanks a lot for the quick reply !
I have not received the sensor yet. I just wanted to be sure that it will be supported at some point by EV3dev. I'll give a try then and let you know.

Thanks

GJpy, 2015-08-19:
Ok I reveived the sensor. It seems that it doesn't show up at all when I plug it on the EV3. Any special setup I should run or something ?

Thanks for your help !

dlech, 2015-08-20:
If pins 2 and 3 are not connected internally, then the sensor won't be automatically detected. You can use the ev3dev user interface (brickman) to manually configure the port. Under *Device Browser > Ports*, select the port that it is connected to. Then press the down button to select the *Set mode* button. Press enter and then change the mode to *nxt-analog*.  Then go back to *Device Browser > Sensors* and you should have a *nxt-analog* device. Select the sensor and then select *Watch values*. The voltage should change when you bend the sensor.

GJpy, 2015-08-20:
It works perfectly. Thanks a lot.

dlech, 2015-08-20:
Awesome! If you figure out the scaling for it, we could make a proper driver for it instead of using the generic `nxt-analog` dirver.

GJpy, 2015-08-21:
Yep, I'm looking at the scaling of the value right now. I'll keep you posted :) cheers

GJpy, 2015-08-23:
Hi,

I remapped the value got from the sensor from MIN 4150 mv ( not bent )=> MAX 4510 mv ( fully bent ) to 0.0 => 1.0 (I called it the "bend" property). It works quite well.
I have also a system of threshold to make the sensor more "sensible". I just reduce the MAX value ( here, 4510 ) according to a percentage property ( clamped to the MIN value of course ).

ALl that perfecly fit my needs :)

dlech, 2015-09-07:
@DexterInd, how does scaling work for the dFlex sensor? Is it linear? Do these values sound about right?

DexterInd, 2015-09-07:
Nothing really sophisticated.  The blocks we wrote for the older NXT were just reading the resistance.  And the resistance changes with bending; it's a simple analog sensor.  I will push the datasheet up to github for public consumption, and the example code in RobotC.

DexterInd, 2015-09-07:
Hey folks, I've dropped in our original code, the part datasheet, and everything else I could find about the dFlex.  https://github.com/DexterInd/DI_LEGO_NXT/tree/master/dFlex

dlech, 2015-09-07:
I've added an actual driver now, which will be included in the next kernel release. This uses the scaling provided by @GJpy. From what I can tell, 4150 mV corresponds to about 5kΩ and 4510 mV corresponds to about 20kΩ.

Hopefully this scaling will work for most people. We might consider adding a calibrate command to the driver in the future, but currently we don't support commands for NXT/Analog sensors, so that will be a separate issue.

dlech, 2015-11-09:
Fix released in kernel cycle 8 (`*-8-ev3dev-*`).

closed 2015-11-09 by dlech