ev3dev/ev3dev #380: Include cdrom and sr_mod modules to allow mode switching of WNDA3200 WiFi stick

cthil, 2015-08-15:
Hi,

for some unknown reason ath9k_htc doesn't mode switch my WNDA3200 WiFi stick as it works on a 3.16.0-4-amd64 box.

I tried usb_modeswitch, but this doesn't work, too. eject'ing the /dev/sr0 device with the eject command however seems a viable alternative, verified on the RasbPi and amd64.

How do I get the sr_mod and cdrom module? With ev3dev, the sg device is present, but not the sr, after inserting the WiFi stick.

System information:
-------------------
ev3-ev3dev-jessie-2015-07-08.img
Linux ev3dev 3.16.7-ckt14-6-ev3dev-ev3 #1 PREEMPT Wed Jul 8 14:08:29 CDT 2015 armv5tejl GNU/Linux
Before switching: Bus 001 Device 015: ID 0cf3:20ff Atheros Communications, Inc. AR7010 (no firmware)
After switching: Bus 001 Device 014: ID 0846:9018 NetGear, Inc. WNDA3200 802.11abgn Wireless Adapter [Atheros AR7010+AR9280]

dlech, 2015-08-15:
It looks like the relevant kernel configuration options is `BLK_DEV_SR`. We can turn it on in the next kernel release. You can compile your own kernel if you want to try it before then.

cthil, 2015-08-15:
Thanks. Is there a release schedule somewhere? At the moment, the only place where I have the toolchain installed is on the brick itself (and, from compiling my own C++ programs there, I know this would be painfully long for the kernel itself).

dlech, 2015-08-15:
The release schedule is just whenever I have some free time. 

Cross-compiling the kernel is fairly easy: <https://github.com/ev3dev/ev3dev-buildscripts>

dlech, 2015-09-07:
Fix has been released in kernel `v3.16.7-ckt16-7-ev3dev-ev3`.

closed 2015-09-07 by dlech