ev3dev/ev3dev #381: EV3 has been booting for 22 minutes

The-Penultimate-Defenestrator, 2015-08-18:
I inserted the microSD and booted for the first time. The getting started page says it's supposed to take longer on first boot, but should it take this long? Is it safe to pull out the card or pull out a battery or will this corrupt files?

dlech, 2015-08-18:
Longer means about 1 minute at most. Yes, it is safe to pull out the card or the battery - although holding the back and enter buttons for about 3 seconds to reset is generally easier than removing the battery.

My guess is that the SD card is already corrupt or defective or too big (> 32GB). Try flashing again or using a different SD card.

The-Penultimate-Defenestrator, 2015-08-18:
SD card is 8 GB, I'll try re-flashing. New ones arrive tomorrow.

The-Penultimate-Defenestrator, 2015-08-18:
FIxed. I had copied the EV3-BOOT from the disk image, but not the other file. Now it works.

closed 2015-08-18 by The-Penultimate-Defenestrator

G33kDude, 2015-08-18:
I don't recall ever having to copy files around to flash my SD cards, it's always just flash them then stick it into the bot and boot.

The-Penultimate-Defenestrator, 2015-08-18:
I flashed it using disk utility on my Mac, but the image had two partitions, only one of which I flashed

rhempel, 2015-08-18:
No, you take the entire image that was downloaded and then:
```
dd if=source.img of=/dev/sdx bs=4M
```
No need to monkey around with the images or partitions in the image or on the SD card.
Not being picky, but what part of the instructions was not clear on that so we can improve it if necessary.

The-Penultimate-Defenestrator, 2015-08-19:
Yes, I didn't do this because it didn't work the first time. This is how I
fixed my issue. Before, I had used Disk Utility