ev3dev/ev3dev #382: apt-add-repository

tiblop, 2015-08-20:
Hello I can't make apt-add-repository to work ! 
I tried to install from : 

sudo apt-get install software-properties-common
sudo apt-get install python3-software-properties
sudo apt-get install python-software-properties

but then it doesn't work... please help

dlech, 2015-08-20:
Did you run `sudo apt-get update` first? Do you have an Internet connection to your EV3?

tiblop, 2015-08-20:
yes and yes

dlech, 2015-08-20:
Hmm... `sudo apt-get install software-properties-common` works for me. Are there any error messages?

tiblop, 2015-08-20:
yes, apt-get install works it is then when i try to use apt-add-repository it says : 
apt-add-repository could not find a distribution template for ev3dev/jessie
I thought it was because i should install apt-add-repository differently

dlech, 2015-08-20:
I see, what is the actual command line you are typing in?

tiblop, 2015-08-20:
sudo apt-add-repository http://ev3dev.org/debian

dlech, 2015-08-20:
That repository is already included in the apt sources on the EV3. Are you trying to install this repository on a regular computer?

tiblop, 2015-08-20:
no but i've done some fail manipulation. at some point it told me that it was a problem with my source so i did a backup of my source and delete everything on it then i deleted the backup by mistake... :/ I try to start all over from scratch but it didn't help ...

tiblop, 2015-08-20:
i'm trying to install ROS by following thie tutorial: 
https://github.com/moriarty/ros-ev3/blob/master/brickstrap-build-status.md
but it messed up my whole system and now i'm stuck.

dlech, 2015-08-20:
It looks like those instructions are supposed to be run on a host computer, not on the EV3 itself.

dlech, 2015-08-20:
Also, that project has not been updated in many months and will likely not work correctly on a recent release of ev3dev.

tiblop, 2015-08-20:
:/ ok so now that explain things... So now how can I fix my EV3?

dlech, 2015-08-20:
You could re-flash the SD card.

dlech, 2015-09-07:
Assuming this issue is resolved. You can reopen if there is still a problem.

closed 2015-09-07 by dlech