ev3dev/ev3dev #383: enabling rndis_wlan in kernel (for USB Wifi Dongle USR5421)

Darko-S, 2015-08-22:
On ev3dev upgraded as of today USB Wifi Dongle USR5421 does not work, but the driver should be present in the kernel.

(EV3DEV) lsusb
Bus 001 Device 002: ID 0baf:011b U.S. Robotics Wireless MAXg Adapter [Broadcom 4320]

(EV3DEV)uname -a
Linux ev3dev 3.16.7-ckt14-6-ev3dev-ev3 #1 PREEMPT Wed Jul 8 14:08:29 CDT 2015 armv5tejl GNU/Linux

I have tested the same dongle on PC with Debian jessie, mainline kernel 3.16.0-4, without having installed any packages named packages rndis* or compat*

(PC) lsmod:
---snip---
rndis_wlan             43358  0 
rndis_host             12698  1 rndis_wlan
cfg80211              405538  1 rndis_wlan
cdc_ether              12564  1 rndis_host
usbnet                 30844  3 rndis_host,rndis_wlan,cdc_ether
usbcore               195340  10 uhci_hcd,rndis_host,rndis_wlan,usb_storage,ehci_hcd,ehci_pci,usbhid,usbnet,cdc_acm,cdc_ether
---snip---

So I believe that driver is part of kernel, as stated also here https://wireless.wiki.kernel.org/en/users/Drivers/rndis_wlan.
Please, could you enable it?

dlech, 2015-08-22:
The will be in the next kernel release, whenever that is.

dlech, 2015-09-07:
Fix has been released in kernel `v3.16.7-ckt16-7-ev3dev-ev3`.

closed 2015-09-07 by dlech

Darko-S, 2015-09-24:
Many thanks, driver works like a charm :)