ev3dev/ev3dev #384: What speeds to expect with the vanilla drivers?

jodigiordano, 2015-08-24:
Sorry for the noob question.

I followed [the tacho-motor tutorial](http://www.ev3dev.org/docs/tutorials/tacho-motors/) and I was wondering what speeds I should expect with this (and all other) drivers.

For example, I did this command:
```
$> time for run in {1..1000}; do cat /sys/class/tacho-motor/motor0/position > /dev/null; done

real 0m40.786s
user 0m7.700s
sys  0m21.100s
```

([Difference between real, user and sys](http://stackoverflow.com/questions/556405/what-do-real-user-and-sys-mean-in-the-output-of-time1) so we're all on the same page)

So it takes 40.7 ms (`0m40.786s / 1000`) to poll the `position` of a motor.
As I see it (please correct me, I'm not good with those things):

* `51.73%` of the time is spent waiting for IO in my command (`0m21.100s`)
* `18.87%` of the time is spent doing other stuff in my command (`0m7.700s`)
* `29.40%` of the time (the rest) is spent by the OS doing other stuff.

Does an average of 40 ms is normal using this technique?

Are there better ways to interact with the drivers than shell commands in the context of a library wrapping the drivers? I'm working on [this Ruby library](https://github.com/jodigiordano/lego_ev3) and I was thinking using file handles instead of system calls, but I feel like I'm missing something...

Thanks!

ddemidov, 2015-08-24:
Caching file handles can help a lot. See ev3dev/ev3dev-lang#25 for some numbers. Also, I think a lot of what you measure here is starting a new process (`cat` in this case).

jodigiordano, 2015-08-24:
Yes, after posting my question, I tried this little Ruby script:
```
def with_timer(&block)
  start = Time.now

  block.call

  finish = Time.now
  diff = finish - start

  puts (diff * 1000).to_i # in ms.
end

print "With file handler... "
with_timer do
  file = File.open('/sys/class/lego-sensor/sensor0/value0')

  (1..1000).each do
    file.rewind
    file.read
  end
end

print "With system call... "
with_timer do
  (1..1000).each do
    `cat /sys/class/lego-sensor/sensor0/value0`
  end
end
```

Results are pretty conclusive:
```
root@ev3dev:~# ruby test.rb
With file handler... 757
With system call... 48606
```

jodigiordano, 2015-08-24:
Thanks @ddemidov, I will implement a solution based on https://github.com/ev3dev/ev3dev-lang/pull/25

closed 2015-08-24 by jodigiordano

rhempel, 2015-08-24:
Strongly suggest you look at the ev3dev-lang repo and use the autogen facility :-)