ev3dev/ev3dev #385: Connection timed out

jurezabkar, 2015-08-26:
Setting a new mode to gyro or ultrasonic sensor, e.g.:

```bash
#!/bin/bash

for sen in /sys/class/lego-sensor/*; do
    cat ${sen}/driver_name
    cat ${sen}/port_name
    for m in $(cat ${sen}/modes); do
        echo ${m}
        echo ${m} > ${sen}/mode
    done
done
```
results in *Connection timed out* or *No such device* error:
```
root@ev3dev:~# ./sensors.sh 
lego-ev3-touch
in4
TOUCH
./sensors.sh: line 8: echo: write error: Invalid argument
lego-ev3-gyro
in1
GYRO-ANG
GYRO-RATE
GYRO-FAS
GYRO-G&A
GYRO-CAL
./sensors.sh: line 8: echo: write error: Connection timed out
lego-ev3-color
in3
COL-REFLECT
COL-AMBIENT
COL-COLOR
REF-RAW
RGB-RAW
COL-CAL
lego-ev3-us
in2
US-DIST-CM
US-DIST-IN
US-LISTEN
US-SI-CM
./sensors.sh: line 8: echo: write error: Connection timed out
US-SI-IN
./sensors.sh: line 8: echo: write error: Connection timed out
US-DC-CM
./sensors.sh: line 8: echo: write error: No such device
US-DC-IN
./sensors.sh: line 8: echo: write error: No such device
```
For color sensor, setting a new mode always works as expected.

ddemidov, 2015-08-26:
See also ddemidov/ev3dev-lang-python#9

dlech, 2015-08-26:
The touch sensor is a different error and a different driver, so it needs to be a separate issue.

IIRC, one or more of the `*-CAL` modes on the UART sensors need the same note as the `IR-S-ALT` mode. Setting those modes causes the sensor to reset itself, which would cause errors like those reported.

However, that shouldn't be happening with the ultrasonic sensor. I'll try to reproduce it when I get a chance.

dlech, 2015-09-06:
I've finally set aside some time to look at this. One thing I should have mentioned is to look at the output of `dmesg` for any hints as to what is going on. This is just a good tip in general.

As expected, for the LEGO EV3 IR sensor `IR-S-ALT` mode, in the kernel log, I saw:

    tty ttyS0: Reconnected due to: No data since last keep-alive.

Likewise, I saw the same error for the ultrasonic sensor. At one point though, I got an unexpected result:

    tty ttySU0: Reconnected due to: Unexpected mode.

I also locked up the kernel a few times.

---

I put this on my logic analyser to see what was going on. I was able to see that it is the ultrasonic sensor that stops responding and not the EV3. So, I think we can safely say that most of the problem is with the sensor and not our drivers.

As was [already pointed out](https://github.com/ddemidov/ev3dev-lang-python/issues/9#issuecomment-134983444), this sensor seems to be a bit touchy when it comes to changing modes, particularly when switching from the `US-LISTEN` mode to another mode (or the same mode).

It can take up to 5 seconds for a UART sensor to renegotiate, so this explains both `write error: Connection timed out` and `write error: No such device`. They just occur at different points during the renegotiation.

So, I'm going to close this as "wontfix" because the problem is really with the sensors. If I can reproduce the lockup or other error that I saw, I'll open a new issue.

closed 2015-09-06 by dlech