ev3dev/ev3dev #386: "Unsupported platform" on apt-get upgrade on Raspberry Pi2b

BertLindeman, 2015-09-06:
Could well be an invalid combination in my `/etc/apt/sources.list`
```
deb http://ftp.debian.org/debian jessie main contrib non-free
deb http://archive.raspbian.org/raspbian jessie main contrib non-free rpi
deb http://ev3dev.org/debian jessie main
deb http://apt.adafruit.com/raspbian/ wheezy main
```
As I use a touchscreen.
If so: sorry beforehand.

Today a lot of updates came in (62), no idea which one caused this.
Result of apt-get upgrade:
```
Setting up initramfs-tools (0.120) ...
update-initramfs: deferring update (trigger activated)
Processing triggers for initramfs-tools (0.120) ...
update-initramfs: Generating /boot/initrd.img-3.18.13-5-ev3dev-rpi2
Unsupported platform.
run-parts: /etc/initramfs/post-update.d//flash-kernel exited with return code 1
dpkg: error processing package initramfs-tools (--configure):
 subprocess installed post-installation script returned error exit status 1
Errors were encountered while processing:
 initramfs-tools
E: Sub-process /usr/bin/dpkg returned an error code (1)
```

result of `cat /proc/cpuinfo` (showing here just one processor)
```
processor       : 0
model name      : ARMv7 Processor rev 5 (v7l)
BogoMIPS        : 38.40
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xc07
CPU revision    : 5
```

Result of `cat /proc/device-tree/model` is `Raspberry Pi 2 Model B Rev 1.1`

Anything else I forgot?

dlech, 2015-09-06:
You are mixing raspbian and debian. This can get confusing because both are "armhf", but raspbian's armhf is compiled for the armv6 instructions set. This will run on both RPi1 and RPi2. Debian's armhf, on the other hand, is compiled for the armv7 instruction set. It will run on the RPi2 only.

Since both are called "armhf", you can't do the multi-arch thing and install programs from either source like you can with i386/amd64 or armel/armhf.

For ev3dev, raspbian is they only supported [package source](https://github.com/ev3dev/brickstrap/blob/master/rpi-ev3dev-jessie/root/etc/apt/sources.list) for RPi1 and debian is the only supported [package source](https://github.com/ev3dev/brickstrap/blob/master/rpi2-ev3dev-jessie/root/etc/apt/sources.list) for RPi2.

BertLindeman, 2015-09-08:
So indeed user error, sorry.
Closing.

Thanks @dlech

dlech, 2015-09-15:
Actually, it turns out this is a bug in the flash-kernel package. Looks like RPi is returning a different value for the model or I just never tried to upgrade a kernel before.

dlech, 2015-09-15:
Fixed in f1c3b628baa275b63f144bdb2065700d747c2505

dlech, 2015-09-16:
Fixed released in `flash-kernel 3.35ev3dev6`.

closed 2015-09-16 by dlech

BertLindeman, 2015-09-17:
@dlech Started on my rpi2 with `rpi2-ev3dev-jessie-2015-09-15.img` 

While installing lxde (nearly 500 packages) ran into:
```
Processing triggers for initramfs-tools (0.120) ...
update-initramfs: Generating /boot/initrd.img-4.1.7-7-ev3dev-rpi2
Unsupported platform.
run-parts: /etc/initramfs/post-update.d//flash-kernel exited with return code 1
dpkg: error processing package initramfs-tools (--configure):
 subprocess installed post-installation script returned error exit status 1
Processing triggers for menu (2.1.47) ...
Processing triggers for systemd (215-17+deb8u2) ...
Errors were encountered while processing:
 initramfs-tools
E: Sub-process /usr/bin/dpkg returned an error code (1)
```

Using sources.list from this image:
```
deb http://ftp.debian.org/debian jessie main contrib non-free

deb http://ev3dev.org/debian jessie main
```

Part of `cat /proc/cpuinfo`   : 
```
processor       : 0
model name      : ARMv7 Processor rev 5 (v7l)
BogoMIPS        : 38.40
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xc07
CPU revision    : 5

```

Where can I verify the string that is used here?
`cat /proc/device-tree/model` shows null-delimited string:
```
Raspberry Pi 2 Model B Rev 1.1
```
Do you want this in another issue?

dlech, 2015-09-17:
Nope, this is the same bug. It looks like somehow I missed getting the fixed flash-kernel into the ev3dev armhf repository. If you do another `sudo apt-get update; sudo apt-get upgrade` and then manually run `sudo flash-kernel`, you should be good to go.

dlech, 2015-09-17:
> Where can I verify the string that is used here?

in `/usr/share/flash-kernel/db/all.db`

BertLindeman, 2015-09-17:
Thanks @dlech , you fixed it again.
The `sudo flash-kernel` might not be needed as the `apt-get upgrade` already installed the new kernel version. But I did it anyway, does not take so much time.
Rebooted OK too ;-)

autocracy, 2016-01-28:
Had to strace to find out that I needed to add
```
Machine: BCM2709
```
into the block with `Machine: Raspberry Pi 2 Model B` et al to stop that error and get a successful flash.

dlech, 2016-01-28:
@autocracy, sounds like you don't have device-tree enabled or are missing dtbs or are not using an ev3dev kernel or didn't install from an ev3dev image file. If you are using an ev3dev image, please open a new issue and provide the information requested at <http://www.ev3dev.org/support/>.