ev3dev/ev3dev #388: add a shellscript to launch apps with proper vt setup

ensonic, 2015-09-14:
In https://github.com/ev3dev/brickman/pull/2 @dlech  suggested to add a script like proposed in https://gist.github.com/ensonic/13a14012a152d976a27b to https://github.com/ev3dev/ev3dev-tools.

Open issues:
1. pass back the tty so that the caller can eventually kill the process
2. the ev3dev-tools git has no package

For 1) we could add another option to let the called pass the tmp-file for the tty-name instead of creating our own. It would be up to the caller to remove the tmpfile once read.

dlech, 2015-09-14:
I'll take care of 2) and that sounds like good suggestion for 1).

In light of some of the other ideas we have been discussing, I am wondering if we should take a step back and ask ourselves if our current setup is actually the best way to go about things.

For example, do we really want all users' programs listed in brickman? We could add a feature where you log into brickman as a user. There would be a "ev3dev" guest account that doesn't require a password (at least in brickman) to keep things simple. This could also help with the issue of simplifying/hiding settings for certain users.

rhempel, 2015-09-14:
What about linking your programs into the ev3dev guest account home directory. Those are the only ones that get listed in the file browser.
I think we should avoid having to force the user to interact very much at all with the front panel.

ensonic, 2015-09-14:
One simplification i wanted to suggest is to skip listing /home in the file browser if all that it contains is /home/user and go to /home/user directly. From a practical standpoint, I think having one user account would work for most of the people. And I think it would be even fine to have it there by default on the images (without a password).

I would expect the following items in the /home/user dir:
* hand-written programs (in c/go/js/py/...)
* generated programs (e.g. those that openroberta will send

We could investigate if we can cram more programs into the screen by using a icon-view.

rhempel, 2015-09-14:
Yes, a passwordless guest account as the single default account is probably fine for most users - not terribly secure against ssh logins though. First boot instructions could ask users to change it - and for bonus points, a pattern of buttion presses on the keypad could be used as an easier password than the virtual (and hard to see) keyboard :-)

ensonic, 2015-09-15:
Actually like root has a default password, we can setup a default one for guest too. Since brickman is started as root, it can always run user programs via su or sudo. And the default password for the user would provide some basic ssh protection.

dlech, 2017-07-25:
I've created a new client/service program called `console-runner` that is shipping with ev3dev stretch. It takes care of this better than a shell script could.

https://github.com/ev3dev/console-runner

closed 2017-07-25 by dlech