ev3dev/ev3dev #390: make install in ev3devKit is not installing all files needed for brickman

ensonic, 2015-09-16:
As a developer I'd like to build from git:

    cd ev3devKit-Build
    cmake ../ev3devKit -DCMAKE_BUILD_TYPE=string:Debug -DEV3DEVKIT_DESKTOP=bool:Yes -DCMAKE_INSTALL_PREFIX:PATH=/usr
    make all && sudo make install
    cd ../brickman-build
    cmake ../brickman -DCMAKE_BUILD_TYPE=string:Debug -DBRICKMAN_TEST=bool:Yes

unfortunately this give me:

    -- checking for modules 'ev3devkit-desktop;gtk+-3.0'
    --   package 'ev3devkit-desktop' not found

When I build ev3devKit with -DBUILD_SHARED_LIBS=bool:Yes I get shared libs, but still the pkgconfig data is not installed.

ensonic, 2015-09-16:
As a workaround, I was trying to build the debian package and install that, but:

    cd ev3devKit
   dpkg-buildpackage -us -uc -d

https://gist.github.com/ensonic/f3773d38fabba3e46756

dlech, 2015-09-16:
We figured this out on IRC. In case anyone else has the same problem, I renamed the "master" branch to "ev3dev-jessie", so pull the remote "ev3dev-jessie" branch to get the latest changes.

closed 2015-09-16 by dlech

ensonic, 2015-09-16:
This was caused by me still using master of ev3devkit instead of ev3dev-jessie.