ev3dev/ev3dev #391: Screen doesn't work from non-root account.

G33kDude, 2015-09-20:
Attempting to use GNU Screen from a non-root account results in the error `Directory '/var/run/screen' must have mode 777.`

It can be fixed until the next reboot by following the instructions in the error message and running `sudo chmod 777 /var/run/screen`, but it'd be nicer if it just worked to start with.

dlech, 2015-09-20:
Debian bug report: https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=471763

Unfortunately, not very useful information there.

dlech, 2015-09-20:
What happens if you add your user to the `utmp` group?

ddemidov, 2015-09-20:
This is off topic, but tmux works on ev3 (and I think it's better than
screen).

G33kDude, 2015-09-20:
@dlech Adding my user to utmp didn't seem to help, but what did was modifying the permissions of `/usr/bin/screen` to match those on my desktop. That is, `chown root:utmp /usr/bin/screen && chmod g+s /usr/bin/screen`

@ddemidov screen holds one specific advantage over tmux, and that is the ability to open terminals on serial devices. For example, you can use it to interact with the NXT while it's running @rhempel's pbLuad firmware.

dlech, 2015-09-20:
So for some reason, the `utmp` group is not getting set as it should in `/etc/init.d/screen-cleanup`.

Ah...

    david@ev3dev:~$ stat -c%a /usr/bin/screen
    2755
    david@ev3dev:~$ systemctl status screen-cleanup
    ● screen-cleanup.service
       Loaded: masked (/dev/null)
       Active: inactive (dead)

`screen-cleanup` is not being run at all.

dlech, 2015-09-20:
Oh, wait, you set the stick bit on `/usr/bin/screen`. That's why it works for you now. I'm not sure that's what we want to do.

G33kDude, 2015-09-20:
Those are the permissions for the screen binary on my laptop, which I have not modified. It was there on it in ev3dev as well before I started messing with the permissions. Changing the user group erases the bit for some reason, so I had to add another command to re-add it.

G33kDude, 2015-09-20:
to be more specific

Xubuntu 14.04:
```
geek@GeekPC:~$ ls -l /var/run/ | grep screen
drwxrwxr-x 3 root       utmp         60 Sep 20 14:47 screen
geek@GeekPC:~$ ls -l /usr/bin/screen
-rwxr-sr-x 1 root utmp 421768 Nov  7  2013 /usr/bin/screen
```

ev3-ev3dev-jessie-2015-07-08:
```
pi@ev3dev:~$ ls -l /var/run/ | grep screen
drwxrwxr-x  3 root  utmp     60 Sep 20 15:23 screen
pi@ev3dev:~$ ls -l /usr/bin/screen
-rwxr-sr-x 1 root root 373660 Oct  8  2014 /usr/bin/screen
```

dlech, 2015-09-20:
It looks like the problem is that `chown` is not working in the brickstrap shell.

dlech, 2015-09-20:
It looks like this is an unimplemented feature in PRoot that we are using as part of brickstrap. I've opened an issue there: <https://github.com/proot-me/PRoot/issues/92>.

Makes me wonder how many other packages are not setup correctly because of this.

dlech, 2015-10-30:
Since proot seems to be abandoned, I have removed the dependency on proot in <https://github.com/ev3dev/brickstrap/commit/8047c6af20d541f67510bd8f45522257bfff53eb>. It turns out that we can do the same thing using kernel namespaces.

dlech, 2015-12-30:
Fix has been released.

http://www.ev3dev.org/news/2015/12/30/ev3dev-jessie-2015-12-30-release/

closed 2015-12-30 by dlech