ev3dev/ev3dev #392: Default user/password

dlech, 2015-09-21:
As @ensonic, @rhempel and I started discussing in #388, we are going to start shipping with a default user and password instead of using root. I have a bit of a dilemma here though in choosing the name.

The obvious choice is `ev3dev`, but when you create a new user, it also creates a new group with that name. However, we already have a system group called `ev3dev` for use with sensors and motors,  so these would conflict.

Then next choices are `ev3` or `dev`. `ev3` is OK, but ev3dev runs on other hardware as well (i.e. Raspberry Pi). `dev` is OK too, but it could be confused with `/dev`.

Another possibility is using the generic `user`. Or we could get creative and think of something else. I would like to keep it short though, because I will probably be typing it frequently.

And we need a password to go with it.

Thoughts? Suggestions?

rhempel, 2015-09-21:
What's the issue with the default username being `ev3dev`? I understand there's a system group that already has the name, but if I create my own user, and make them a member of the `ev3dev` group I don't have any more special permissions at the user level of the `ev3dev` user files.
Is the issue simply avoiding having system and group usernames that are the same?

dlech, 2015-09-21:
> Is the issue simply avoiding having system and group usernames that are the same?

Right, you can't have 2 groups with the same name. I guess we could make the users default group `ev3devuser` for example. We could use the `adduser --ingroup` option to set the group to the `ev3dev` system group, but this could cause some unexpected results, or maybe not, I don't know.

dlech, 2015-09-21:
I'm looking and `/etc/adduser.conf` has an option to disable usergroups, which would take care of this problem. I'm thinking that is what I will do. All users will have the default group of `users`.

Still need a password though...

rhempel, 2015-09-21:
what about just having a user called `guest` and ensuring that the `motd` for that user indicates that it's an insecure account.

laurensvalk, 2015-09-28:
For beginning users, I think having a default user with password and permissions set up properly, would be ideal. In my opinion having no default password actually complicates things. (Whenever prompted for a password, I find entering a password more intuitive than pressing Enter or leaving a field blank, depending on the situation.)

Basically such that all/most steps in  "Step 5: First things to do with ev3dev" such as setting up a user and adding it to the appropriate groups can be skipped.

Then the only thing that needs to be done after writing the SD card is "Step 4: Setup a network connection". In some cases, that might prevent having the user to do any command line typing at all. (Say, if in the future we have something like Jupyter or Open Roberta installed by default.) 

Some more ideas for user names, including some short ones:
lego
robot
ev3
lmsuser
ev3user //shorter than ev3devuser
ev3rstorm
ev3programmer
beginner

And passwords. Most of the above, or:
mindstorms
linux4mindstorms

ensonic, 2015-09-28:
As far as I see it, most users will never ever have to enter the password for the user. I'd like to have the 'user' pregenerated:
* for having a home-directory to store generated/downloaded programs
* having a restricted user to run such code

With usernames/passwords I'd avoid the lego brandnames. What about: username=user and password=ev3dev

rhempel, 2015-09-28:
I'd avoid using `user` as a username because it can introduce confusion in documentation:

```
foo_cmd -verbose -u user
```

Wait, do we put in `user` or the actual username? My personal feeling is `guest` is good enough - and as Stephan says, using LEGO brandnames is to be avoided.

Password can be `ev3dev`, or some other phrase - it would be really good if the `motd` could be tuned for `guest` to encoureage them to create a new user.

dlech, 2015-09-28:
I don't like  `guest` because to me that invokes the idea of a limited user account and our default account will be fully privileged (e.g. sudo powers). And I was able to work around the technical issues with having `ev3dev` as the user name. So, unless someone has a really good argument otherwise, I am going to go with

    user: ev3dev
    password: robot

for the next release.

G33kDude, 2015-09-28:
I use the raspbian defaults on my ev3, since it makes it much simpler to copy/paste from guides written for the raspberry pi.

    user: pi
    password: raspberry

That being said, this is not a raspberry pi and ev3dev/robot sounds like as good credentials as any.

laurensvalk, 2015-09-28:
> So, unless someone has a really good argument otherwise, I am going to go with user: ev3dev, password: robot

User ev3dev being a member of ev3dev group seems somewhat ambiguous to me, even if it's technically possible. You could flip it to get:

> user: robot
> password: ev3dev

rhempel, 2015-09-28:
I'm OK with using `ev3dev` as the default user - I understand the conflict with the `ev3dev` system group, but it seems like the least bad alternative.

@laurensvalk has a good suggestion as well, I'm OK with either one

dlech, 2015-09-28:
OK, `robot:ev3dev` it is.

closed 2015-09-28 by dlech

dlech, 2015-12-29:
I made a last minute change on the password. It is now `robot:maker`.

rhempel, 2015-12-29:
I like it! :+1:

WasabiFan, 2015-12-29:
Yeah -- `maker` is much easier to type than both `ev3dev` and `r00tme` (and admittedly I regularly made the mistake of mistyping `r00tme` because of the "ohs")

cho934, 2016-03-24:
Do you keep an root connexion ? What is the root default password ?

dlech, 2016-03-24:
By default, `root` does not have a password configured (so you cannot log in directly as root). We recommend to use `sudo` with the `robot` user instead.

josefmatzes, 2016-05-14:
Ark my Projects I had to Open The file var/lib but this file can't ne opened Geom The User Robot and with The User Robot Comes a error (13:permissions dennied can someone give me a hint?

dlech, 2016-05-14:
As I said in my previous comment, use [sudo](https://help.ubuntu.com/community/RootSudo) when you need root privileges.

heidashuai, 2017-02-20:
Sorry,but I met a problem with the user groups. I wanted to use sound function so I add robot to audio group,but I find that robot is out of ev3dev group!
And I do not know how to make it back,since I have no right to do the operations...

ensonic, 2017-02-20:
@heidashuai  Audio should already work for the 'robot' user. What did you try to do and how did it fail? Also what commands did you run to change the groups?

heidashuai, 2017-02-20:
@ensonic Thank you.
I use `usermod -G audio robot` to modify the group,but after this script my robot only belongs to audio group and no longer the sudoer member. 
I have already modified the group file by plugin the sd-card to another pc. Now my robot belongs to sudoer and ev3dev groups again.But I wonder if there are other groups I should add?

dlech, 2017-02-21:
`usermod -a -G ...` will add instead of replace groups. However, if the robot user was not already in the `audio` group, there are other problems.