ev3dev/ev3dev #393: Changes for upcoming release

dlech, 2015-09-21:
I'm using this issue to keep track of my notes for changes for the next release of ev3dev.

-[ ] <http://www.ev3dev.org/docs/tutorials/connecting-to-the-internet-via-usb/> needs to be updated for the new ev3-usb.service and changes to brickman.
- [ ] <http://www.ev3dev.org/docs/getting-started/> needs to be updated for new default user and password (#392).
- [ ] Release notes or announcement need to list manual upgrade process.

dlech, 2015-09-21:
Manual upgrade process:

    * Make the following changes to `/etc/adduser.conf`:
        * Change `USERGROUPS=yes` to `USERGROUPS=no`.
        * Uncomment the line that starts with `#EXTRA_GROUPS=`. Then remove `users` and add `ev3dev input bluetooth i2c`.
        * Uncomment the line `#ADD_EXTRA_GROUPS=1`.
    * If you created a user called `ev3dev` already, you don't have to do anything else.
    * If you have created a non-root user already (other than `ev3dev`):
        * Log in as `root`.
        * Run the following command where `<old_username>` is replaced by your old user name:
                usermod -l ev3dev -g users -m -d /home/ev3dev <old_username>
    * If you haven't created a non-root user yet, run the following commands:
                adduser --gecos "" ev3dev
                usermod -a -G sudo ev3dev

dlech, 2015-10-30:
Hey, @rhempel and @ensonic and anyone else that want something included in the next release...

This release is going to have some major changes and there is a lot more to keep track of. I have create a [milestone](https://github.com/ev3dev/ev3dev/milestones/ev3dev-jessie%202015-Q4) to keep track of everything that needs to get included and so that I don't forget to mention it in the release notes.

If there is something that you want included but is not on the milestone, please comment in that issue. If the issue is in another repository, please create a corresponding issue in <https://github.com/ev3dev/ev3dev/issues> so that we can add it to the milestone.

dlech, 2015-10-30:
You can use [this search](https://github.com/ev3dev/ev3dev/issues?utf8=%E2%9C%93&q=is%3Aopen+milestone%3A%22ev3dev-jessie+2015-Q4%22+-label%3A%22fix+committed%22) to see what is left to do.

dlech, 2015-12-16:
@rhempel, I think getting ev3dev-lang-python updated with the latest kernel changes is the last major thing to do before an image release (other than documentation changes, which are very major). I think event2sensor will have to wait unless you think you will have it packaged and ready to go sometime this week.

rhempel, 2015-12-16:
OK, I'll be tight for time to get event2sensor done by Christmas - so ev3dev-lang-python will be now. Right now it's set up as an external to  ev3dev-lang based on my repo - it's OK to leave it that way?

dlech, 2015-12-16:
> Right now it's set up as an external to ev3dev-lang based on my repo - it's OK to leave it that way?

Yes. (in my opinion, that is the preferred way to have it).

WasabiFan, 2015-12-16:
@dlech Why are the manual instructions crossed out if users need to follow it (it's linked to from the [announcement](http://www.ev3dev.org/news/2015/12/15/Package-Release/))? I assume there's a reason for it, but I can't seem to figure it out.

dlech, 2015-12-16:
Because of https://github.com/ev3dev/brickstrap/commit/8047c6af20d541f67510bd8f45522257bfff53eb. All existing image files are fundamentally broken (permission issues). In the next release, we will have to ask everyone to reflash their SD card to fix this.

WasabiFan, 2015-12-16:
Ahhh I see. Thanks for explaining.

rhempel, 2015-12-16:
And we may need to help them get any files in their `/home/robot` or `/home/username` saved.

closed 2015-12-30 by dlech