ev3dev/ev3dev #394: Improve the dialog classes in ev3devKit

ensonic, 2015-09-21:
Now sure, how much we'd like to model this after GtkDialog and also sync this with MessageDialog. 
* MessageDialog has a title and uses border_bottom = 1 to draw a separator under the title.
  * InputDialog could also get a title + message, where the title is getting the separator
* MessageDialog has no dialog buttons (like "Okay") which makes sense to save space

In Gtk (https://developer.gnome.org/gtk3/stable/GtkDialog.html) a dialog has 3 areas: header, content and action.
If we want to harmonize the dialogs, we can probably move more code up to the baseclass (e.g. the title).

Having general access to the content area would allow apps to style the message. But thats probably better done with api on the MessageDialog to access the message_label.

dlech, 2015-09-21:
One other thing I have been wanting to improve is how to make the text input easier/more obvious. For example, currently sometimes if you press the enter button on a text input widget, it will open the on-screen keyboard and sometimes it will just edit locally.

My thought is to make the input widget include a keyboard icon on the right side, then all inputs can either be edited locally or use the on-screen keyboard.

dlech, 2015-12-09:
Can this issue be closed or is there still something left to do?

closed 2015-12-10 by ensonic