ev3dev/ev3dev #395: Uknown usb device after reboot

TomHartogs, 2015-09-22:
After following each and every "get started" step I managed to succesfully launch the ev4dev terminal in Putty. However, after rebooting the brick the usb device doesn't show up in the network section of device manager anymore and i'm unable to connect the through Putty. I already reinstalled once before which solved the problem untill I had to reboot again.. did anyone else experience this or does someone have a solution to solve this?

ensonic, 2015-09-22:
Did you plug the USB cable afte the brick booted (not sure t it matters under windows).

TomHartogs, 2015-09-22:
I kept it plugged in at first but also tried unplugging and replugging it. Rebooted the device several times by now but so far no succes

dlech, 2015-09-22:
Did you run `apt-get update` and `apt-get upgrade` before your rebooted?

Do you have another way to connect to your EV3, such as Bluetooth or Wi-Fi?

TomHartogs, 2015-09-22:
Yes I did, succesfully updated and rebooted afterwards. Currently at school it's difficult to connect through the wifi but i'll give this a try at home or I will try pairing with bluetooth. I assume it's best to perform another clean install before trying this?

ensonic, 2015-09-22:
I think something broke with the recent updates. On the ev3 CDC keeps saying Inactive. On my linux machine I get this in syslog when connecting via usb:

    Sep 22 17:06:02 square.roam.corp.google.com kernel: [453588.406748] usb 3-1: device not accepting address 39, error -62
    Sep 22 17:06:02 square.roam.corp.google.com kernel: [453588.806841] usb 3-1: new high-speed USB device number 41 using xhci_hcd
    Sep 22 17:06:07 square.roam.corp.google.com kernel: [453593.808159] xhci_hcd 0000:00:14.0: Timeout while waiting for address device command
    Sep 22 17:06:12 square.roam.corp.google.com kernel: [453599.013542] xhci_hcd 0000:00:14.0: Timeout while waiting for address device command

dlech, 2015-09-22:
The change in question is this: https://github.com/ev3dev/ev3-systemd/commit/1af7d53447d9823a9bfac616231f8602fea49e30

Of course, it is working great for me on Windows, Linux and Mac, so I can't reproduce these problems.

It is expected that RNDIS and CDC on the USB menu don't work any more after upgrade. This menu will be going away in the next release of brickman. There should now be a ev3-usb.service that is always on and provides both RNDIS and CDC ECM.

If you have an alternative way to connect to your EV3, please check the output of `sudo journalctl -xn` and `sudo systemctl status -l ev3-usb.service`.

ensonic, 2015-09-22:
https://gist.github.com/ensonic/30d92906f6f125d78cce

after I restrted the service it connected.

dlech, 2015-09-22:
Must be a race condition between `ExecStartPre=/sbin/modprobe libcomposite` and the `/sys/kernel/config/usb_gadget` being created.

dlech, 2015-09-22:
I was able to reproduce the problem by unplugging all of the sensors and motors from my EV3. So, as a temporary workaround, you can plug in a bunch of sensors and motors and reboot and it will probably work.

TomHartogs, 2015-09-22:
Thanks allot for the quick response! At home right now I don't have a Brick but i'll give it a go tomorrow at school first thing in the morning. I will keep you updated.

dlech, 2015-09-22:
This should be fixed in `ev3-systemd 4.0.1`. Hopefully you can get to the point where you can `apt-get update; apt-get upgrade` and get the new package.

bblais, 2015-09-24:
I tried "you can plug in a bunch of sensors and motors and reboot and it will probably work." and the CDC choice is still inactive.  I'll try reimaging the sd card if nothing else works.

dlech, 2015-09-24:
If you have done `apt-get upgrade` after 19 Sept. 2015, it is expected that the USB menu in brickman no longer works. The whole menu will be removed in the next release of brickman. The USB connection should "just work" regardless of what the USB menu says. Are there any wired connections shown under *All Network Connections*? Do you have another way to connect to the EV3?

ensonic, 2015-09-24:
I had to do 'apt-get dist-upgrade' for it.

dlech, 2015-09-28:
If anyone else can confirm that this fixes the "unknown USB device" problem, we can close this issue.

TomHartogs, 2015-09-29:
I just tried plugging in a bunch of sensors but this didn't help for me. The choice did stay active even after rebooting but for some reason I wasn't able to connect to the internet. So I tried a full reinstall again which didn't solve anything so I was stuck without an internet connection. I decided to download the latest zip file from github but with this file on the sd card the brick keeps booting to the original software instead of the ev3dev software. I'm gonna give it another try with the older version, walking through the entire "getting started".

TomHartogs, 2015-09-29:
After the reinstall I get the first issue again, i'm connected and able to reboot without any problem. Putty will just start after the reboot however I am not able to update. I'm able to ping googles public DNS (8.8.8.8) and other ip adresses but other than that there seems to be no connection. But I'm guess there will be a solution for this problem somewhere else

dlech, 2015-09-29:
It sounds like DNS is not working. I would suggest disabling Internet Connection Sharing in Windows and the re-enable it to reset it.

TomHartogs, 2015-10-06:
I manually changed the DNS server which made me able to update. This did bring me back to the Putty error, even with a bunch of sensors plugged in. By using the "2015-09-13" file will I miss out on allot of important updates? I'm considering just using this version without upgrading since it seems to work fine that way. We are using the Brick for a school project which will last 20 weeks (week 5 currently) so i'm able to give it another go in a few weeks aswell.

dlech, 2015-10-06:
Yes, it is OK to keep working without upgrading, especially if you are constrained for time.

Plugging in sensors and motors was only meant to fix the "unknown device in device manager" error, not any other errors.

TomHartogs, 2015-10-13:
Maybe I should post this under a differrent Issue but i'm curious on how I should compile and run a program on the brick using C++. While reading the read me I found out that building on the EV3 itself should be fairly easy. However when I try to install the build-essentials I get a message telling me I should maybe run apt-get update. But doing this will make me have to deal with the USB error again. Is there anyway around this or should I cross compile? 

Apologies in advance if some questions seem fairly stupid. I'm an IT student who just started his second year, before all this I didn't have any experience. Even though I really what i'm doing right now I still find it very hard.

ddemidov, 2015-10-13:
I think you should go for cross-compiling anyway, because compiling C++ on the brick is extremely slow. This [wiki page](https://github.com/ev3dev/ev3dev/wiki/Using-brickstrap-to-cross-compile-and-debug) worked for me.

dlech, 2015-12-16:
Fix was released some time ago, but there has been some confusion due to brickman not being updated. Please open a new issue if you have any more problems similar to this one.

closed 2015-12-16 by dlech