ev3dev/ev3dev #396: Linked Rubik's Cube Solver does now work on current release of ev3dev

akhayyat, 2015-09-23:
The first project linked in the _"What can it do?"_ section in the home page of the ev3dev.org website is the impressive Rubik's Cube Solver. It's an incredibly cool project, and a very effective and nice way to get people's attention to look further into the ev3dev project, which is what happened to me.

Unfortunately, after spending many hours building the robot and trying to get the software to work, I discovered that that code does not run on the current ev3dev release (issue submitted for the Rubik's Cube Solver project here: https://github.com/cavenel/ev3dev_examples/issues/4).

For this to be effective, either that project gets fixed, or it should be removed from the homepage to avoid misleading people.

dwalton76, 2015-09-23:
Funny timing, I noticed the same thing earlier this week.  I forked it and will send a pull request once I am finished (hopefully later this week).  My fork is https://github.com/dwalton76/ev3dev_examples

cavenel, 2015-09-23:
Thank you for your work @dwalton76! I will merge all that as soon as you pull it :)

For the original issue, it is an interesting question: are projects linked in the "What can it do?" section supposed to be easily doable by others? I did the rubiks cube solver mainly for fun at the very beginning of ev3dev and as a proof of concept. I didn't do it thinking that other people would copy it.
But then I did the Print3rBot in a way that anyone could redo it at home easily. It is two very different approaches and maybe it should be specified somewhere on the project page. In the same way, I guess that @CallumCameron's Dalek is not meant to be easily done, but @donwojtallo's snake is.

dlech, 2015-09-24:
> are projects linked in the "What can it do?" section supposed to be easily doable by others?

They are mostly just projects I thought were interesting examples of what people have done with ev3dev. Perhaps I should make it clear that these don't necessarily come with instructions. :smile:

> ...and maybe it should be specified somewhere on the project page.

You just have to send a pull request for that.

dwalton76, 2015-09-28:
FYI this is working now

rhempel, 2015-09-28:
Yes, thanks @dwalton76 - we need all the help we can get to keep the examples current.

cavenel, 2015-09-28:
Awesome work @dwalton76! Thank you. Everything is merged, but I didn't have time to try myself. Trusting you 100% here.
Will try that soon!
I would say that next step is to make it run on @ddemidov python bidings... But maybe not really necessary if everything works without!

rhempel, 2015-09-28:
My pure Python bindings got a test at MakerFaire this weekend - the DrawString bot worked flawlessly for 1.5 days - and used the motor and sensor bindings for the IR sensor. Next step - finish testing the rest of the API

dwalton76, 2015-09-28:
@cavenel is the idea to replace your ev3.py Robot class with @ddemidov python bindings?

dlech, 2015-09-28:
Sounds like this has been fixed, so closing this issue. We are also working on some updates to the website which removes the part that has mislead you.

closed 2015-09-28 by dlech

dwalton76, 2015-10-13:
Hey @cavenel I have one more pull request outstanding for the rubiks solver...can you take a look?  I need to take the robot apart for a Halloween project :)

cavenel, 2015-10-14:
Thank you for this outstanding pull request! And for all the work done on this rubiks cube solver!
Is it ok if I credit you somewhere on the project page for all the work done?

I did not test all changes but I will as soon as I have 2 hours!

PS: I had the nicest comment about this robot on Youtube last week: "141 seconds? That's 64 times slower than Cube Twister, the cube solver running mainly EV3 Basic, which can solve cubes in 2.2 seconds. HaHaHa. See EV3Basic.com." :-D

dwalton76, 2015-10-14:
yep, crediting on the project page is fine :)

cube twister is amazing...he/she must be offloading some of the CPU intensive parts to a server though. If not I cannot imagine how it crunches the colors or figures out the solution so quickly.

dwalton76, 2015-10-21:
I have blog where I (very rarely these days) write about my lego projects.  I put up a quick entry on the Rubiks cube robot running ev3dev:
http://programmablebrick.blogspot.com/2015/10/mindcuber-ev3-in-python.html