ev3dev/ev3dev #397: CDC USB choice always inactive even when selected

bblais, 2015-09-24:
I've been working on the ev3 for a little while, no problem.  Last night I did a number of software updates (mostly python) - this may be irrelevant.  This morning, the CDC choice under networking says "(inactive)" even when choosing it, and thus I can't connect at all to the device from my Mac.  The device does not show up (anymore) on the list of devices.  The image I'm using is ev3-ev3dev-jessie-2015-09-13.  I did all of the apt-get updates.

I'm not sure how to even debug this, because I can't gain any access to the device, because the CDC is always inactive.

ensonic, 2015-09-24:
This is most likely a dupe of #395 .

dlech, 2015-09-28:
Not exactly a dup, but the same explanation applies that the USB menu is going away and should be ignored if you have upgraded to `ev3-systemd 4.0.1`.

Check with:

```
$ dpkg-query -W ev3-systemd
ev3-systemd	4.0.1
```

bcorey, 2015-11-09:
I've got the same problem - I followed the "Getting Started" instructions from the ev3dev website and everything worked very well until I turned the ev3 off. 
Now when I turn it on the RNDIS box says "Inactive" even when checked, the "Wireless and Networks" section of the Brickman menu says "Status: No connections", and the "Wired" category does not show up under "All network connections". The computer recognizes that a USB device called "EV3+ev3dev" is plugged in, but the "Hardware" tab inside the device's "Properties" window says "Device status: this device cannot start". Pinging the ev3 from the command prompt returns nothing.

I've read through both this and #395, and understand that this is a bug with the current version of Brickman, but is there anything I can do until the next version comes out that will make this work?

dlech, 2015-11-09:
The "Inactive RNDIS box has nothing to do with your issue. Please open a new issue for

> The computer recognizes that a USB device called "EV3+ev3dev" is plugged in, but the "Hardware" tab inside the device's "Properties" window says "Device status: this device cannot start". Pinging the ev3 from the command prompt returns nothing.

bcorey, 2015-11-09:
Here's #427. So if the RNDIS is not related, do I have two issues here?

dlech, 2015-11-09:
The USB screen in brickman is going away, so the fact that it does not work is not a problem. It is supposed to be that way.

PLeVasseur, 2015-11-30:
I followed along with this [wiki page](https://github.com/ev3dev/ev3dev/wiki/Using-brickstrap-to-cross-compile-and-debug) page about cross-compiling and the [tutorial](http://www.ev3dev.org/docs/tutorials/connecting-to-the-internet-via-usb/) on connecting to the internet via USB, but I am not able to ssh into the brick from Ubuntu 14.04. When I used the [image released on 09-13] (https://github.com/ev3dev/ev3dev/releases/tag/ev3-ev3dev-jessie-2015-09-13), I was able to SSH into the brick.

When running off the image I created through brickstrap, the CDC selection is always Inactive, even when selecting it. When running off the 09-13 released image, the CDC selection will toggle between Inactive and Active.

I took down some notes during the process that I got an error during the step on the wiki to install build-essentials. I repeated the install of build-essentials and this time received no error. I don't know if that's related, but I thought I'd mention it in case it is.

```
Errors were encountered while processing:
 systemd
 systemd-sysv
 init
E: Sub-process /usr/bin/dpkg returned an error code (1)
```

```
$ ssh root@10.42.0.3 -v
OpenSSH_6.6.1, OpenSSL 1.0.1f 6 Jan 2014
debug1: Reading configuration data /etc/ssh/ssh_config
debug1: /etc/ssh/ssh_config line 19: Applying options for *
debug1: Connecting to 10.42.0.3 [10.42.0.3] port 22.
debug1: Connection established.
debug1: identity file /home/peter/.ssh/id_rsa type -1
debug1: identity file /home/peter/.ssh/id_rsa-cert type -1
debug1: identity file /home/peter/.ssh/id_dsa type -1
debug1: identity file /home/peter/.ssh/id_dsa-cert type -1
debug1: identity file /home/peter/.ssh/id_ecdsa type -1
debug1: identity file /home/peter/.ssh/id_ecdsa-cert type -1
debug1: identity file /home/peter/.ssh/id_ed25519 type -1
debug1: identity file /home/peter/.ssh/id_ed25519-cert type -1
debug1: Enabling compatibility mode for protocol 2.0
debug1: Local version string SSH-2.0-OpenSSH_6.6.1p1 Ubuntu-2ubuntu2.3
debug1: Remote protocol version 2.0, remote software version OpenSSH_6.7p1 Debian-5
debug1: match: OpenSSH_6.7p1 Debian-5 pat OpenSSH* compat 0x04000000
debug1: SSH2_MSG_KEXINIT sent
Read from socket failed: Connection reset by peer
```

```
$ ping 10.42.0.3
PING 10.42.0.3 (10.42.0.3) 56(84) bytes of data.
64 bytes from 10.42.0.3: icmp_seq=1 ttl=64 time=0.543 ms
64 bytes from 10.42.0.3: icmp_seq=2 ttl=64 time=0.522 ms
64 bytes from 10.42.0.3: icmp_seq=3 ttl=64 time=0.571 ms
^C
--- 10.42.0.3 ping statistics ---
3 packets transmitted, 3 received, 0% packet loss, time 1999ms
rtt min/avg/max/mdev = 0.522/0.545/0.571/0.027 ms
```

I ran the command given above, within the brickstrap shell:
```
$ dpkg-query -W ev3-systemd
ev3-systemd	5.0.0
```

dlech, 2015-12-01:
My previous comments still apply. There has not been a release of brickman since then. This is currently the expected behavior.

dlech, 2015-12-16:
Fix has been released in brickman 0.7.0. More casual users will want to wait for the next image release before updating. See the [announcement](http://www.ev3dev.org/news/2015/12/15/Package-Release/) for more info.

closed 2015-12-16 by dlech