ev3dev/ev3dev #398: naming mixup for usr/share/{brickman,ev3devkit} ?

ensonic, 2015-09-24:
As part of my changes on the brickman UI I added two icons to brickman. One hack that I had to do to make my two new icons load: ln -s /usr/share/brickman/ /usr/share/ev3devkit

Even though that most of the images are in ev3devkit in git, on the device the are stored under /usr/share/brickman/ and that is where I install my new images too, but Ev3devKit.Ui.Icon.create_context_from_png(filename) tries to find them in /usr/share/ev3devkit

dlech, 2015-09-28:
I'm not seeing the same behavior:

```
root@ev3dev:~# dpkg-query -W -f Version brickman
Versionroot@ev3dev:~# dpkg-query -W brickman
brickman	0.6.2
root@ev3dev:~# ls /usr/share/brickman/
splash.png
root@ev3dev:~# dpkg-query -W libev3devkit-0.1-0
libev3devkit-0.1-0	0.2.1
root@ev3dev:~# ls /usr/share/ev3devkit/
bluetooth12x12.png  bluetooth-connected7x9.png  lock5x6.png  usb12x12.png  wifi12x12.png  wifi-idle12x9.png
bluetooth7x9.png    ethernet12x12.png           lock7x9.png  usb7x9.png    wifi12x9.png   wps9x9.png
```

To load an image in brickman, you need to specify the full path. There is a `PKGDATADIR` constant defined in [Version.vala](https://github.com/ev3dev/brickman/blob/ev3dev-jessie/src/Version.vala.in) for this. So, for example at <https://github.com/ev3dev/brickman/pull/3/files#diff-620acda0609c822cfd8deb428eed3fa8R38>, I would put this instead:

    file = Path.build_filename (PKGDATADIR, "openroberta-connected12x9.png");

ensonic, 2015-09-29:
$ dpkg -L libev3devkit-0.1-0
/.
/usr
/usr/share
/usr/share/doc
/usr/share/doc/libev3devkit-0.1-0
/usr/share/doc/libev3devkit-0.1-0/copyright
/usr/share/doc/libev3devkit-0.1-0/changelog.gz
/usr/lib
/usr/lib/libev3devkit-0.1.so.0.2.1
/usr/lib/libev3devkit-0.1.so.0

but I found it:
$ dpkg -S /usr/share/ev3devkit
ev3devkit-data: /usr/share/ev3devkit
$ dpkg -L ev3devkit-data
/.
/usr
/usr/share
/usr/share/ev3devkit
/usr/share/ev3devkit/bluetooth7x9.png
/usr/share/ev3devkit/lock5x6.png
/usr/share/ev3devkit/usb7x9.png
/usr/share/ev3devkit/wifi-idle12x9.png
/usr/share/ev3devkit/wifi12x9.png
/usr/share/ev3devkit/ethernet12x12.png
/usr/share/ev3devkit/usb12x12.png
/usr/share/ev3devkit/bluetooth12x12.png
/usr/share/ev3devkit/wifi12x12.png
/usr/share/ev3devkit/bluetooth-connected7x9.png
/usr/share/ev3devkit/lock7x9.png
/usr/share/ev3devkit/wps9x9.png
/usr/share/doc
/usr/share/doc/ev3devkit-data
/usr/share/doc/ev3devkit-data/copyright
/usr/share/doc/ev3devkit-data/changelog.gz

ensonic, 2015-09-29:
Thanks for pointing out  PKGDATADIR. It works now.

closed 2015-09-29 by ensonic