ev3dev/ev3dev #399: Unable to get wifi to start on boot on image rpi-ev3dev-jessie-2015-09-15.img.zip

antonvh, 2015-09-25:
This is my setup:
```
anton@ev3dev-rpi1:~$ cat /etc/wpa_supplicant/wpa_supplicant.conf 
# ctrl_interface=/var/run/wpa_supplicant
network={
       ssid="SSID"
       psk="****"
}

network={
	ssid="fabrique-gast"
	psk="Leukdatjeerbent!"
}

anton@ev3dev-rpi1:~$ cat /etc/network/interfaces
# interfaces(5) file used by ifup(8) and ifdown(8)
# Include files from /etc/network/interfaces.d:
# source-directory /etc/network/interfaces.d

auto lo
iface lo inet loopback

auto wlan0
allow-hotplug wlan0
iface wlan0 inet dhcp
    wpa-conf /etc/wpa_supplicant/wpa_supplicant.conf

allow-hotplug eth0
iface eth0 inet dhcp
```

After boot I get this annoying message 5 times. I have to wait for these messages to stop before I can log in.
```
cfg80211: Calling CRDA to update world regulatory domain
```

Then I have to do this to get wifi working :(
```
$ifdown wlan0
$ifup wlan0
```

I get some Job for smbd.service failed error on the way. But after that wlan is up...

dlech, 2015-09-25:
We are using [ConnMan](https://wiki.archlinux.org/index.php/Connman) for network management, so this type of network configuration is ignored.

Follow the link to learn how to use the `connmanctl` command to configure Wi-Fi from the command lne.

antonvh, 2015-09-27:
Thanks! Works. It's a bit confusing that connman doesn't connect to wifi automatically when eth0 is connected.

dlech, 2015-09-27:
Wi-Fi can't connect automatically without a passphrase unless it is an open network, in which case it would be a security risk to connect automatically.

You can close this issue if your problem is solved.

antonvh, 2015-09-27:
What I meant is that connman connects to wifi just fine after booting, fully automatically. But ONLY if there is no eth0. That confused me. Nu need to enter a passphrase each time btw, with `connman agent on`.

closed 2015-09-27 by antonvh