ev3dev/ev3dev #400: How to run with sudo from the File Browser?

dwalton76, 2015-09-25:
I am using an updated ev3.py from ev3dev_examples, one snag I am hitting is that I have to be root to write to the motor stop_command, etc files.  When I am sshed in this is fine but how can I run this with sudo from the File Browser on the ev3?

dwalton76@ev3dev:~/ev3dev_examples/python$ ./rubiks_test.py 
2015-09-25 16:16:03,203 rubiks_test.py     INFO: import complete
Traceback (most recent call last):
  File "./rubiks_test.py", line 14, in <module>
    rub = Rubiks()
  File "/home/dwalton76/ev3dev_examples/python/pyev3/rubiks.py", line 22, in __init__
    self.mot_push = Motor('A', 'flipper')
  File "/home/dwalton76/ev3dev_examples/python/pyev3/ev3.py", line 168, in __init__
    self.stop()
  File "/home/dwalton76/ev3dev_examples/python/pyev3/ev3.py", line 384, in stop
    self.set_stop_mode(stop_mode)
  File "/home/dwalton76/ev3dev_examples/python/pyev3/ev3.py", line 274, in set_stop_mode
    self._write_file('stop_command', value)
  File "/home/dwalton76/ev3dev_examples/python/pyev3/ev3.py", line 179, in _write_file
    self.write(path, str(value))
  File "/home/dwalton76/ev3dev_examples/python/pyev3/ev3.py", line 29, in write
    with open(path, 'w') as pout:
IOError: [Errno 13] Permission denied: u'/sys/devices/platform/legoev3-ports/lego-port/port4/outA:lego-ev3-l-motor/tacho-motor/motor0/stop_command'
dwalton76@ev3dev:~/ev3dev_examples/python$

dlech, 2015-09-25:
You don't need to be root. you just need to be a member of the `ev3dev` group.

G33kDude, 2015-09-25:
Isn't the user brickman uses to run files already part of the ev3dev group as well?

dlech, 2015-09-25:
brickman currently uses the owner of the file to run the file (but that will change in the near future). If you created the user account that is the owner of the file, then that account is only a member of the `ev3dev` group if you added that group to the account.

dwalton76, 2015-09-26:
Ah makes sense now. Thanks guys I will give this a try tomorrow.

dwalton76, 2015-09-28:
ok the only remaining snag is reading /dev/input/by-path/platform-gpio-keys.0-event , I need sudo to read it (I am a member of the ev3dev group):

dwalton76@ev3dev[python]# ./foo.py 
Traceback (most recent call last):
  File "./foo.py", line 46, in <module>
    main()
  File "./foo.py", line 33, in main
    with open('/dev/input/by-path/platform-gpio-keys.0-event', 'r') as fd:
IOError: [Errno 13] Permission denied: '/dev/input/by-path/platform-gpio-keys.0-event'
dwalton76@ev3dev[python]# 

foo.py is the "Polling Key States" from https://github.com/ev3dev/ev3dev/wiki/Using-the-Buttons

G33kDude, 2015-09-28:
I think you have to be a member of the input group for that. The standard command to add you to all the normal user groups (as listed in the Getting Started guide) is `usermod -a -G sudo,ev3dev,plugdev,audio,video,input,bluetooth,i2c dwalton76`

dwalton76, 2015-09-28:
That worked, thank you :)

closed 2015-09-28 by dlech

ArielBas, 2016-04-22:
Hello 

I am VERY new to Ev3, Ev3dev as well as python, but I need it for a university project. I am trying to run Laurens Valk’s segway.py file (for the BALNC3R robot).  He has helped me to the point where I now have my Ev3 connected to the network and I was able to drag and drop the files to the Ev3 file browser (using FileZilla). I have also installed Python 3.5 64 bit version if that is of any relevance, and I have executed the necesssary update commands I found to be able to import the ev3dev-lang-python binding with 
>> from ev3dev.auto import * 

But I am not even 100% sure if this was realy necessary for what I want to do. 
From there I had trouble starting the program using the Ev3 buttons, so Laurens suggested using the ./segway.py command – but it doesn’t work. I have now executed every update/ungrade command I have come across, I have made the file executable, and I have also installed dos2unix which solved similar problems for other people by setting 
$ chmod u+x filename.py
$ dos2unix hello.py
$ ./filename.py 
(where I used ‘segway’ instead of ‘filename’)

I have also installed Python 3.5 64 bit version if that is of any relevance, and I have executed the necesssary update commands I found to be able to import the ev3dev-lang-python binding with 
>> from ev3dev.auto import * 

But I am not even 100% sure if this was realy necessary for what I want to do. 
I am using Windows 10, and used the ev3-ev3dev-jessie-2015-12-30.img file on my SD card. I have, however, not tried changing anything on Laurens’ segway.py file content as I don’t think the problem would be there. This is the error message I keep getting after  $ ./filename.py 
 
_robot@ev3dev:~$ dos2unix segway.py
dos2unix: converting file segway.py to Unix format ...
robot@ev3dev:~$ ./segway.py
Traceback (most recent call last):
  File "./segway.py", line 35, in <module>
    subprocess.call(['./makelinks.sh'])
  File "/usr/lib/python3.4/subprocess.py", line 537, in call
    with Popen(*popenargs, **kwargs) as p:
  File "/usr/lib/python3.4/subprocess.py", line 859, in __init__
    restore_signals, start_new_session)
  File "/usr/lib/python3.4/subprocess.py", line 1457, in _execute_child
    raise child_exception_type(errno_num, err_msg)
PermissionError: [Errno 13] Permission denied_

Can anyone please help me with this? I see it is the same error number as the error talked about earlier in this chat, is it the same problem? And if so, how can I fix the membership?
Also, is there a particular sequence of commands I need to execute every time, or will ./segway.py always work after the first time? And lastly, if ./segway.py works once, will I also be able to run the file directly from the ev3 file browser without the USB connection?

harmooni, 2016-04-22:
@ArielBas 
Your python script can't execute makelinks.sh, probably because it is not marked es executable (similar to your problem with segway.py). Try setting chmod u+x makelinks.sh

dlech, 2016-04-22:
> Also, is there a particular sequence of commands I need to execute every time, or will ./segway.py always work after the first time?

Once you have the permissions set correctly on the file, you don't need to perform this sequence of steps each time. And actually there are some things you can do to prevent the need for these steps in the first place. 

1. When editing files, make sure you use a text editor that is aware of line endings and set it to use Unix (LF) line endings. For example, Notepad++ is a popular text editor for Windows that can do this.

    [image]

2. When copying files using Filezilla, you can set the executable bit in Filezilla.

    [image]
    [image]

> And lastly, if ./segway.py works once, will I also be able to run the file directly from the ev3 file browser without the USB connection?

Yes, once you have the executable bits set correctly (and line ending problems fixed if there was a problem with that), then you can run the file from the file browser on the EV3 without having to use a remote terminal.

ArielBas, 2016-04-26:
Thank you Harmooni. That seems to have solved the problem I had. Now I just need to adjust the code a little to do what we need it to do.

And thank you dlech. I tested some output from the command line as well as the Ev3 brick and it works for both, just like you said it would. 

Really appreciate both of your feedback. 
Enjoy the rest of your week.