ev3dev/ev3dev #401: No i2c bus on brickpi ev3dev?

antonvh, 2015-09-27:
When I booted with the original Dexter industries BrickPi image this worked, on ev3dev brickpi it doesn't. I don't know where to start looking. Pointer please?
```
root@ev3dev-rpi1:~# i2cdetect -y 1
Error: Could not open file `/dev/i2c-1' or `/dev/i2c/1': No such file or directory
```
(I installed i2ctools and rebooted. nothing i2c-like in the /dev/ direvtory.)
No sensors are attached, but the voltage regulator is on the i2c bus, and I want to read battery voltage from it.

dlech, 2015-09-27:
The Dexter Industries image is based on an older version of Rasbian and the Raspberry Pi firmware. ev3dev on the other hand is rather cutting edge at the moment (assuming you are using the latest version - please post the info requested [here](http://www.ev3dev.org/support/) so we know exactly which version you are running).

We are using device overlays, which are controlled by `config.txt`. You can read more about it [here](https://www.raspberrypi.org/documentation/configuration/device-tree.md). And you can find `config.txt` at `/boot/flash/config.txt` (this location is unique to the ev3dev distribution).

I think you should be able to get it turned on by making the appropriate changes to `config.txt`. If you figure it out, let us know and we will update the default image so that it is enabled by default. Or even better, open a pull request for the files [here](https://github.com/ev3dev/brickstrap/blob/master/rpi-ev3dev-jessie/root/boot/flash/config.txt) and [here](https://github.com/ev3dev/brickstrap/blob/master/rpi2-ev3dev-jessie/root/boot/flash/config.txt).

dlech, 2015-09-28:
I got to looking at this and i2c should be enabled by the brickpi overlay already. It worked in the previous release, so I'm not sure what has broken. I made a change in the commit above, but it doesn't seem to fix the problem.

dlech, 2015-09-28:
OK, found the problem, the `i2c-dev` module is not loading. Either we need to compile this into the kernel or add the module to a list somewhere to load it automatically.

You can get `/dev/i2c-1` working by running `sudo modprobe i2c-dev`.

dlech, 2015-09-28:
> No sensors are attached, but the voltage regulator is on the i2c bus, and I want to read battery voltage from it.

Can you point me to the docs on this? I would like to write a proper power_supply driver for this.

antonvh, 2015-09-29:
Thanks a lot! Modprobe does the trick. I wrote a power supply driver in python already. It's here:
https://gist.github.com/antonvh/c81c247fc03029a1ba6a

And it works!
```
anton@ev3dev-rpi1:~$ sudo modprobe i2c-dev
anton@ev3dev-rpi1:~$ su -
root@ev3dev-rpi1:~# i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- 48 -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --                         
root@ev3dev-rpi1:~# git clone https://gist.github.com/c81c247fc03029a1ba6a.git
Cloning into 'c81c247fc03029a1ba6a'...
remote: Counting objects: 27, done.
remote: Compressing objects: 100% (18/18), done.
remote: Total 27 (delta 8), reused 0 (delta 0), pack-reused 0
Unpacking objects: 100% (27/27), done.
Checking connectivity... done.
root@ev3dev-rpi1:~# cd c81c247fc03029a1ba6a/
root@ev3dev-rpi1:~/c81c247fc03029a1ba6a# python get_voltage.py 
12.34422
```

antonvh, 2015-09-29:
I added a line with `i2c-dev` to `/etc/modules-load.d/modules.conf` for now. I suppose with the next release of the rpi ev3-dev image this won't be necessary anymore.

closed 2015-09-29 by antonvh