ev3dev/ev3dev #402: Resetting the gyro

jurezabkar, 2015-09-27:
I've been trying to reset the gyro in ev3dev but haven't found a way to do so.
Every time I start the program (see below), gyro's value is 0, but I
can't figure out how to reset it inside the program.
```python
from ev3dev import *
from mindstorms_widgets import mindstorms_widgets

gs = gyro_sensor();
assert gs.connected
gs.mode = 'GYRO-ANG'

robot = mindstorms_widgets()
robot.connect_motor( 'left' )
robot.connect_motor( 'right' )

power = 15
robot.move_tank( 'on', lr_power=[power,-power] )
# gs.value() ==0 here
while gs.value()<90:
    pass
robot.move_tank( 'off', brake_at_end=True )
print gs.value() # gs.value() == 90 here (approx.)

# I thought this would reset the gyro, but it doesn't
gs = gyro_sensor();
assert gs.connected
gs.mode = 'GYRO-ANG'

print gs.value() # gs.value() == 90 here
```

dlech, 2015-09-27:
To reset the gyro, you need to change it to a different mode and then back to the angle mode. Also, you don't need to create a new gyro_sensor object. So instead of 

```
gs = gyro_sensor();
assert gs.connected
gs.mode = 'GYRO-ANG'
```

try this...

```
gs.mode = 'GYRO-RATE'
gs.mode = 'GYRO-ANG'
```

dlech, 2015-09-27:
I forgot to mention, that the robot needs to be perfectly still when you do this, otherwise it will mess up the calibration of the sensor. If this is not possible, you could keep track of an "angle offset" in your program and make your own reset function.

jurezabkar, 2015-09-29:
Have you considered having a reset function for the gyro? Switching modes does not seem a natural way to reset it.

dlech, 2015-09-29:
Switching modes actually calibrates the gyro in addition to reseting the angle to zero. This is how you have to calibrate the gyro in the official LEGO software as well. So, that may not be the best choice if your robot is moving.

The alternative is to record current value in an `offset` variable and use that to determine the angle. Due to the way EV3/UART sensors work, it doesn't really make sense to try to implement a reset command at the driver level. I think it would be best to add this at the library level. It looks like you are using <https://github.com/topikachu/python-ev3>, so I would suggest opening an issue there.

closed 2017-07-25 by dlech