ev3dev/ev3dev #403: ESC button is always 'pressed'

dwalton76, 2015-09-28:
Using https://github.com/ev3dev/ev3dev/wiki/Using-the-Buttons "Polling Key States" the ESC button is always shown as pressed:

dwalton76@ev3dev[python]# ./foo.py 
UP (103): released
DOWN (108): released
LEFT (105): released
RIGHT (106): released
ENTER (28): released
ESC (1): pressed
dwalton76@ev3dev[python]#

I am not pressing the ESC button...does this happen for anyone else?

dlech, 2015-09-28:
The "back" button is `BACKSPACE`, not `ESC`. (It was ESC in the very early days of ev3dev.)

dlech, 2015-09-28:
And there is another `ioctl` that tells you if a button is actually present or not. All buttons that are not present will show as pressed.

dwalton76, 2015-09-29:
Ack.  The docs say it is ESC thus the confusion.

dlech, 2015-09-29:
This is on our list of "HELP WANTED" for updating the docs: https://github.com/ev3dev/ev3dev/issues/153#issuecomment-93602107

closed 2015-09-29 by dlech