ev3dev/ev3dev #404: two new package for the repo

ensonic, 2015-09-28:
I've submitted debian packaging for
1)  python-ev3dev: https://github.com/ddemidov/ev3dev-lang-python
2) openrobertalab: https://mp-devel.iais.fraunhofer.de/code/projects/ORA/repos/robertalab/browse/EV3MenuEv3dev

python-ev3dev:
https://drive.google.com/open?id=0B1Upu1QWS-qgenllY2xUMTZPTlU

openrobertalab:
https://drive.google.com/open?id=0B1Upu1QWS-qgenllY2xUMTZPTlU

dlech, 2015-09-28:
Have you read <http://www.ev3dev.org/docs/devtools/packaging-for-ev3dev/>?

rhempel, 2015-09-28:
I had to get my head around the Debian packaging for the Hiawatha package - once you get it (and have @dlech help with fixing mistakes) it's not too bad. It's better to get this formal process in place now so we're not chasing issues later - plus it improves our chances of getting some of these packages into the official Debian upstream later - for example the WeDo kernel modules.

ensonic, 2015-09-29:
I am currently building these from pypi tarball releases. Instead of havign the debian repo inside the python-lang-ev3dev and the openroberta repo, we could also add a 'debian' git repo to the ev3dev project and add the debian build stuff there. Another candidate for the could be the new ev3dev-web-remote @G33kDude is working on.

rhempel, 2015-09-29:
I hope we can include the pure-python bindings I'm working on - @ddemidov and I need to decide on the proper naming for the bindings.
In my opinion, we should:
- Prefer pure language bindings
- Strive to use `autogen` as much as possible to generate the binding file(s)
- Put them in the ev3dev-lang repo using the language name
- Put non-pure bindings in a separate folder with a distinguishing name

I am really impressed with the ease of setting up and using `autogen`!
-

ensonic, 2015-09-29:
@rhempel sure, especially if pure-python is a drop-in replacement we can switch what is in python-ev3dev by default.

ddemidov, 2015-09-29:
C++ based python bindings could be placed into `python-ext`. The only advantage over pure-python I see is the speed, so another possibility could be `python-fast`.

Is it possible in debian to create mutually exclusive packages? So that `python-ev3dev[-pure]` and `python-ev3dev-ext` could only be installed one at a time? Then users who need performance could `apt-get install python-ev3dev-ext` which would handle dependencies and uninstall `python-ev3dev[-pure]` first.

ensonic, 2015-09-29:
@ddemidov: yes packages can have a Conflicts: field, so that one can only install (a) or (b). There are also virtual packages. That is we can make python-ev3dev a virtual package and both python-ev3dev-{pure,ext} implementing it. This way apps can just depend on python-ev3dev and the dependency is satisfied if either of them is installed.

dlech, 2015-09-29:
If you decide that you want separate repositories for debian packaging, just let me know and I will create them.

ensonic, 2015-09-29:
As I just learned, all we need is Provides: and Conflicts: fields. So if we aggree on the names: python-ev3dev-{pure,ext}, I can reflect that in the packages (that is also add packaging to python-ev3dev-pure).

dlech, 2015-09-29:
I would suggest the package names `python-ev3dev` for the "pure" version and `python-ev3dev-fast` (with Provides: python-ev3dev) for the C++ version. This way the "pure" bindings will be installed by default and advanced users can choose to install the "fast" version.

dlech, 2015-09-29:
If neither one is named `python-ev3dev`, I'm not sure how apt would know which one to install.

rhempel, 2015-09-29:
I'm not sure I'd like the name of the other binding to be `python-ev3dev-fast` - the distinguishing feature is that is uses the Boost C++ library, not that it's faster (even if it is faster :-))
We'll also need a test suite to establish API compatibility between `python-ev3dev` and `python-ev3dev-xyz`

dlech, 2015-09-29:
Makes sense. Call it `python-ev3dev-boost` then.

ddemidov, 2015-09-29:
I wonder if better performance of `python-ev3dev-{ext,boost,fast}` could be achieved by just implementing the base `Device` class as a C++ extension. Then the choice between `pure/boost` versions could be made by switching `Device` implementation, and we could join forces with @rhempel on the rest of the bindings. This would also make sure there is no code divergence.

dlech, 2015-09-29:
From the benchmarks [here](https://github.com/ev3dev/ev3dev-lang/issues/3#issuecomment-136130783), the "pure" python can read faster than most of the drivers can update the data that is being read, so I don't really see a need to make it any faster. Maybe just drop the boost option altogether.

ddemidov, 2015-09-29:
If that is the case, I don't mind replacing `boost` version with @rhempel's version.

dlech, 2015-09-29:
For a lack of a better place, I'm going to mention this here:

The "right way" to get framebuffer info in python is [this](https://github.com/LinkCareServices/cairotft/blob/master/cairotft/linuxfb.py). I'm thinking we should probably package it up as a stand-alone package and then it can be used in @rhempel's python library and by anyone else that needs it.

It needs to be cleaned up a bit first and re-licensed, which seems to be compatible with the current license. :stuck_out_tongue_winking_eye:

rhempel, 2015-09-29:
I'll see if I can integrate it into the `pure-python` area - I think all language bindings need a way to interact with the frame buffer

dlech, 2015-11-11:
@ensonic, I have updated your original comment so we can check of progress. There is now a debian package available on ev3dev.org for rhempel's `python-ev3dev 0.5.0`. Let me know when you have the openroberta package updated and I will get it published too.

ensonic, 2015-11-11:
Thanks everybody! I'll rebuild the openroberta package today.

dlech, 2015-12-30:
At long last, there is actually an openrobertalab package in the ev3dev.org repo and it will be included by default in the 2015-Q4 image release.

closed 2015-12-30 by dlech

rhempel, 2015-12-30:
Hey, congratulations! Great to see this in the next release - hopefully we'll see some more traction in the educaiton world for ev3dev!

ensonic, 2015-12-30:
@dlech Thanks a lot!