ev3dev/ev3dev #405: PiCamera on Rpi Ev3dev rpi

antonvh, 2015-09-29:
I plugged in my picamera, because robots with cameras rock! (Here's on I made before https://www.youtube.com/watch?v=DJ6H5ldCbkE) Alas, I couldn't access the picamera on ev3dev jessie. 

This is what I got:
```
anton@ev3dev-rpi1:~$ raspistill -v -o test.jpg

raspistill Camera App v1.3.8

Width 2592, Height 1944, quality 85, filename test.jpg
Time delay 5000, Raw no
Thumbnail enabled Yes, width 64, height 48, quality 35
Link to latest frame enabled  no
Full resolution preview No
Capture method : Single capture

Preview Yes, Full screen Yes
Preview window 0,0,1024,768
Opacity 255
Sharpness 0, Contrast 0, Brightness 50
Saturation 0, ISO 0, Video Stabilisation No, Exposure compensation 0
Exposure Mode 'auto', AWB Mode 'auto', Image Effect 'none'
Metering Mode 'average', Colour Effect Enabled No with U = 128, V = 128
Rotation 0, hflip No, vflip No
ROI x 0.000000, y 0.000000, w 1.000000 h 1.000000
mmal: mmal_vc_component_create: failed to create component 'vc.ril.camera' (1:ENOMEM)
mmal: mmal_component_create_core: could not create component 'vc.ril.camera' (1)
mmal: Failed to create camera component
mmal: main: Failed to create camera component
mmal: Camera is not enabled in this build. Try running "sudo raspi-config" and ensure that "camera" has been enabled
```

of course there is no `raspi-config` on ev3dev... any other way to enable it?

Python package isn't available either...
```
sudo apt-get install python picamera
Reading package lists... Done
Building dependency tree       
Reading state information... Done
E: Unable to locate package picamera
```

dlech, 2015-09-30:
Since there is no `raspi-config`, you have to do it the old fashioned way of editing the configuration file yourself. You can find `config.txt` in `/boot/flash/` (which is unique to ev3dev). See [this page](http://elinux.org/RPiconfig#Camera) for what exactly you need to add there.

As for the `picamera` package, it looks like that is actually the name of the source package. The binary packages (the ones you install) are `python-picamera` and `python3-picamera`. However, they are probably only available from the Raspberry Pi foundation package repository. We purposely don't included this because we do things a bit different and it would break stuff.

I would suggest just install `picamera` with `easy_install` instead.

    sudo apt-get install python-setuptools
    sudo easy_install picamera

antonvh, 2015-10-03:
Works! Ev3dev rocks! Thanks.
I added `start_x=1` after doing `sudo pico /boot/flash/config.txt` and then `sudo reboot`
`raspistill -o cam.jpg` works like a charm.

same with python lib after `sudo easy_install picamera`

closed 2015-10-03 by antonvh