ev3dev/ev3dev #406: print3rbot 40T gear problem

ihsan-akin, 2015-09-29:
The 40T gears, printed or original, are stuck and then ticks on the center 4540797: Beam R. Frame. You can see from the picture. 
To solve this issue, I'll need the print and cut the R Frame.

[image]

Any other suggestions?

dlech, 2015-09-30:
Is this the printerbot by @cavenel?

ihsan-akin, 2015-09-30:
Yes, 
http://www.ev3dev.org/projects/2015/05/06/EV3-Print3rbot/

But not he extension.

cavenel, 2015-09-30:
It would maybe be easier to open an issue directly on the Print3rbot project page: https://github.com/cavenel/ev3-print3rbot/issues
Unfortunately I don't have the original 40T gears to test, didn't have the problem on my printed version, and have a 1 day old baby to take care of ;) 3 reasons that make it difficult for me to solve the issue right now.
I know that others have built the robot with original gear without problem, see https://github.com/cavenel/ev3-print3rbot/issues/2

rhempel, 2015-09-30:
Congratulations on the new arrival - see you in 16 years :-)

cavenel, 2015-10-19:
Isn't 16 a bit too old to play Lego? :-p

Any news on this problem? I see a lot of other people building the robot without complaining about this, so it would be interesting to know if there is indeed a problem here.

closed 2015-12-12 by dlech