ev3dev/ev3dev #407: Samples do not work

burf2000, 2015-09-30:
Just got the latest EV3Dev, updated etc

root@ev3dev:~# echo run-forever > $MC/command
-bash: /sys/class/tacho-motor/motor2/command: No such file or directory

What ever command I run I get
/sys/class/tacho-motor/motor2/duty_cycle_sp: No such file or directory

dlech, 2015-09-30:
Are there any devices listed in the device manager on the EV3?

You can also see a list of all motors by running `ls /sys class/tacho-motor/`.

burf2000, 2015-10-01:
Yes 2 Ev3 motors (plus 2 pf motors connected but not listed)

dlech, 2015-10-01:
The device manager will tell you the name `motorX` which translates to `/sys/class/tacho-motor/motorX`. Since you only have two motors showing up, my guess is that they are `motor0` and `motor1`, not `motor2

dlech, 2015-10-01:
Dang. Accidentally posted somehow. Continuing...

...not `motor2` as in your original comment. Also, power functions motors can't be automatically detected, which is why they are not showing up. You have to manually set the mode of the port. You can do this on the brick in the *Device Manager* under *Ports* or programmatically with `/sys/class/lego-port/portX`.

rhempel, 2015-10-01:
Dave, you can edit your posts :-) I do it all the time.

G33kDude, 2015-10-01:
Ah, but it doesn't send an email notification about the edit does it?

rhempel, 2015-10-01:
Maybe not, but if you catch the edit within a couple of minutes, how likely is that the post has been read...

G33kDude, 2015-10-01:
Email notifications send the body of the comment with the notification, so if someone's following the thread via email, pretty much 100%

dlech, 2016-04-12:
Motor drivers have had a major overhaul since this issue was opened. Please open a new issue for any similar problems.

closed 2016-04-12 by dlech

sudharsh, 2016-04-14:
I have a similar issue where none of the run-* command work. However am able to query the speed of the motor without problems. I guess this means that the driver was loaded successfully.

Is there a way I can debug this?

The `state` of the motor seems to be empty as well.

dlech, 2016-04-14:
As I mentioned in my previous comment, please open a new issue instead of continuing to post on this closed issue. Also, we can't help you if you don't tell us exactly what you tried.

sudharsh, 2016-04-14:
Ok. Will do that