ev3dev/ev3dev #408: What packages are safe to remove to get the smallest possible install size?

grasmanek94, 2015-10-04:
So I want to squeeze out every byte of an install in ev3dev (bytheway, cool project! Just stumbled across it after our teacher said EV3 runs linux that is completely configurable etc.). I am aware that you try to give the most to everyone, and that's great! But there are people like me who just need the bare minimum :) 

I have installed ev3dev, ssh'd into it and there are many packages, the size is about 1.4 GB, but what is the bare minimum to connect with actually?(only wifi and a program writen in C++ to control the motors and read sensors is needed)

 What packages are safe to remove?

I would think as a starting point these command could be run:

    apt-get remove '.*python.*' 'node-.*' 'nodejs.*' 'samba.*' 'lua.*' 'libv8.*' '.*ruby.*'

Or would this cause the system to be unable to boot correctly?

Dry-running this command gives the following as output:

        The following packages will be REMOVED:
          dh-python gyp libjs-node-uuid libpython-stdlib libpython2.7
          libpython2.7-minimal libpython2.7-stdlib libpython3-stdlib
          libpython3.4-minimal libpython3.4-stdlib libruby2.1 libsmbclient
          libv8-3.14-dev libv8-3.14.5 lua5.2 node-abbrev node-ansi
          node-ansi-color-table node-archy node-async node-block-stream
          node-combined-stream node-cookie-jar node-delayed-stream node-forever-agent
          node-form-data node-fstream node-fstream-ignore node-github-url-from-git
          node-glob node-graceful-fs node-gyp node-inherits node-ini
          node-json-stringify-safe node-lockfile node-lru-cache node-mime
          node-minimatch node-mkdirp node-mute-stream node-node-uuid node-nopt
          node-normalize-package-data node-npmlog node-once node-osenv node-qs
          node-read node-read-package-json node-request node-retry node-rimraf
          node-semver node-sha node-sigmund node-slide node-tar node-tunnel-agent
          node-underscore node-which nodejs nodejs-dev nodejs-legacy npm python
          python-crypto python-dnspython python-gi python-gobject python-gobject-2
          python-ldb python-minimal python-ntdb python-pkg-resources python-pyudev
          python-samba python-smbus python-talloc python-tdb python2.7
          python2.7-minimal python3 python3-minimal python3-smbus python3.4
          python3.4-minimal ruby ruby2.1 rubygems-integration samba samba-common
          samba-common-bin samba-dsdb-modules samba-libs smbnetfs
        0 upgraded, 0 newly installed, 96 to remove and 0 not upgraded.

rhempel, 2015-10-04:
I used to always want to build a minimal install too, but then I realized that giving a minimal distribution and then requiring users to add stuff just made it harder for everyone.
We have supplied a pretty decent distribution with a selection of scripting languages and a host independent method of connecting to the EV3.
Having these "extras" does not slow down your robot or make it take longer to boot - except maybe some of the `samba` and `nfs` stuff.
Please consider leaving the distribution as it is and building robots and writing programs to control them, then worry about trimming down the distribution.

rhempel, 2015-10-04:
And please also consider not writing a control program in C++ because then you need to have a C++ compiler on your host - get started with a scripting language and move up from there if yoiu run out of horsepower :-)

dlech, 2015-10-04:
It's safe to remove pretty much any package. Debian packages can be flagged as `Essential` and `apt` will refuse to remove them, so you don't have to worry about removing something really critical. It is probably a bad idea to remove any package with `ev3` in the name though.

But, like @rhempel said, there is not much point in removing a bunch of packages this unless you are just doing it for the fun of it. If you are trying to get more available RAM, you can control which programs are started automatically using `systemctl` rather than uninstalling the packages. For example, you could tell `brickman`, `connman`, `bluez`, `avahi-daemon`, etc. to not start.

dlech, 2015-10-04:
I suppose if you are doing data-logging, you might want more free space on your SD card. In that case, `golang` is another big one.

grasmanek94, 2015-10-04:
Thanks for the comments, that clears much of my questions. Also the task within our project is to write everything in C++ (and I'm somebody who uses C++ for nearly a decade, both on linux/unix-based and windows systems) so this is really our only (and anyway, probably the best) option :) 

So conclusion:

- No need to delete anything, preferably use systemctl
- if really needed to free up space then delete non essential and not containing '.*ev3.*' (and probably the firmware) packages

Thanks!

dlech, 2016-01-28:
Just in case anyone else comes across this issue, you can use our brickstrap tool to create a minimum debian image (similar to [this one](https://github.com/ev3dev/brickstrap/commit/21122acd4db3718b141f60a296377c401eb85a70)). Also, if #416 ever gets implemented, that would be an option too.

closed 2016-01-28 by dlech