ev3dev/ev3dev #409: bootscripts timing out?

ensonic, 2015-10-05:
All packages up-to-date. Once I configurred a wifi network it boot and connects, but before the wifi is 'not available' tired a few boot and the symptoms are on of the two below. Once I've connected via usb, I could restart wpa_supplicant.service (in the case were only that one failed) and connect.

-- 1 ---------------------------------------------------------------------------
systemctl | grep failed
● ev3-bluetooth.service                                                                      loaded failed failed    LEGO MINDSTORMS EV3 built-in Bluetooth
● systemd-rfkill@rfkill0.service                                                             loaded failed failed    Load/Save RF Kill Switch Status of rfkill0
● wpa_supplicant.service                                                                     loaded failed failed    WPA supplicant

● systemd-rfkill@rfkill0.service - Load/Save RF Kill Switch Status of rfkill0
   Loaded: loaded (/lib/systemd/system/systemd-rfkill@.service; static)
   Active: failed (Result: exit-code) since Sat 2000-01-01 01:26:50 CET; 15 years 9 months ago
     Docs: man:systemd-rfkill@.service(8)
  Process: 327 ExecStart=/lib/systemd/systemd-rfkill load %I (code=exited, status=1/FAILURE)
 Main PID: 327 (code=exited, status=1/FAILURE)

Jan 01 01:26:50 ev3dev systemd-rfkill[327]: Failed to get rfkill device 'rfkill0': No such file or directory
Jan 01 01:26:50 ev3dev systemd[1]: systemd-rfkill@rfkill0.service: main process exited, code=exited, status=1/FAILURE
Jan 01 01:26:50 ev3dev systemd[1]: Failed to start Load/Save RF Kill Switch Status of rfkill0.
Jan 01 01:26:50 ev3dev systemd[1]: Unit systemd-rfkill@rfkill0.service entered failed state.

● wpa_supplicant.service - WPA supplicant
   Loaded: loaded (/lib/systemd/system/wpa_supplicant.service; disabled)
   Active: failed (Result: exit-code) since Sat 2000-01-01 01:27:26 CET; 15 years 9 months ago
  Process: 379 ExecStart=/sbin/wpa_supplicant -u -s -O /run/wpa_supplicant (code=exited, status=255)
 Main PID: 379 (code=exited, status=255)

Jan 01 01:26:57 ev3dev systemd[1]: Starting WPA supplicant...

-- 2 ---------------------------------------------------------------------------
systemctl | grep failed
● wpa_supplicant.service                                                                     loaded failed failed    WPA supplicant

dlech, 2015-10-05:
I think there might be two separate issues here. I've seen the bluetooth problem on the Sept. 2015 image. I haven't been able to make it repeatable though. Most of the time it works for me. The problem basically stems from the fact that so much is going on at the same time during boot that it takes too long for `hciattach` to complete.

You can see in [ev3-bluetooth.service](https://github.com/ev3dev/ev3-systemd/blob/ev3dev-jessie/systemd/ev3-bluetooth.service) where I have made an effort to work around this. It was enough in past releases, but apparently is not enough anymore.

The rfkill errors are a result of the first bluetooth error. Once we fix bluetooth, they will go away.

I think the wpa_supplicant error might be a separate issue, but I don't use wi-fi on a regular basis, so I haven't come across this yet. It is possible that it may be related to the other issues, such as depending on rfkill.

ensonic, 2015-10-06:
I made a modification to the wpa_supplicant.service:
```
[Service]
Type=dbus
BusName=fi.epitest.hostap.WPASupplicant
ExecStart=/sbin/wpa_supplicant -u -O /run/wpa_supplicant
StandardOutput=journal
StandardError=journal
```
Basically sensing the logs directly to journal instead of going through syslog. This has the advantage that journalctrl -b -u wpa_supplicant.service actually works (and likewise systemctl status -l wpa_supplicant.service). These are the details I get:
```
Jan 01 01:14:48 ev3dev systemd[1]: Starting WPA supplicant...
Jan 01 01:15:22 ev3dev wpa_supplicant[380]: dbus: Could not request service name: org.freedesktop.DBus.Error.NoReply Did not receive a reply. Possible causes include: the remote application did not send a reply, the message bus security policy blocked the reply, the reply timeout expired, or the network connection was broken.
Jan 01 01:15:22 ev3dev wpa_supplicant[380]: Failed to initialize wpa_supplicant
```

So either dbus is busy or not up yet?

dlech, 2015-10-06:
I wonder if the bluetooth/rfkill issues have something to do with wpa_supplicant not working by tying up dbus and causing a timeout. I wonder if there is a way to tell systemd "ok, stop everything else while we run ev3-bluetooth.service". That would probably fix it.

ensonic, 2015-10-06:
FYI: https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=800615
I added the `After=dbus.service` in the `[Unit]` section and rebooted n times, but this does not fix it.

Regarding your sugestion, I ping the folks in irc, lets see if they have some idea.

ensonic, 2015-10-06:
btw. did you ever try increasing one of these:
/etc/systemd/system.conf:#DefaultTimeoutStartSec=90s
/etc/systemd/system.conf:#DefaultTimeoutStopSec=90s
/etc/systemd/user.conf:#DefaultTimeoutStartSec=90s
/etc/systemd/user.conf:#DefaultTimeoutStopSec=90s

dlech, 2015-10-06:
According to the systemd.service man page:

> Behavior of dbus is similar to simple; however, it is expected that the daemon acquires a name on the D-Bus bus, as configured by BusName=. systemd will proceed with starting follow-up units after the D-Bus bus name has been acquired. Service units with this option configured implicitly gain dependencies on the dbus.socket unit. This type is the default if BusName= is specified.

As far as I can tell, dbus.socket is essentially the same dbus.service, so it is not surprising that adding a dependency does not make a difference. I'm sure that dbus is running, just not responding.

Re: DefaultTimeoutStartSec=90s, I haven't tried that, but the timeout for hciattach in ev3-bluetooth.service is only 30 seconds, so I don't think systemd is killling it.

ensonic, 2015-10-20:
I have turned off swap on my ev3s to short the boot times and curiously since then it never happened again. Of course this is not the solution.

dlech, 2015-10-20:
I have noticed that the swap file stuff does seem to take up a bit of time at boot. Maybe we could find a different way of doing it. Currently we are using `dphys-swapfile`, which is what RPi uses. The RPi with ev3dev boots in a matter of a < 10 seconds, so I'm not sure why it takes so much longer on the EV3.

We can't just completely remove swap though, or trying to apt-get some common packages will fail because of not enough memory.

Showing startup errors is a good idea, but it would be better if we just didn't have any errors to report.

dlech, 2015-12-30:
This release (http://www.ev3dev.org/news/2015/12/30/ev3dev-jessie-2015-12-30-release/) gets rid of dphys-swapfile. Boot time is noticeably faster. Please see if this fixes the problems you are seeing here.

antonvh, 2016-03-15:
My ev3dev image stopped booting. I suspect bluetooth as brickman/bluetooth settings said before it stopped booting:
```
Powered []
Visible []
???
```
Now ev3dev is infinitely stuck on the penguin image. This is the second SD card this happened to. I only use bluetooth to connect.

antonvh, 2016-03-15:
Rebooting the mac and brick helps. Still it would be more elegant if this didn't happen. This BT stuff is a pain to debug.

WasabiFan, 2016-03-15:
@antonvh This doesn't sound like it is a direct continuation of the previous discussion in this issue, so please open a new issue for your specific problem.

dlech, 2016-03-16:
Bluetooth not starting is #572. Brick not booting is new issue.

dlech, 2016-04-12:
One thing I have noticed is that this happens more often if many EV3/UART sensors are attached. In particular the Gyro sensor which seems to use high CPU because of high data rates. In the `3.16.7-ckt26-10-ev3dev-ev3` kernel, I have offloaded EV3/UART sensor data processing to a separate kernel thread. I don't know if this will make a difference or not, but pay attention when you upgrade to this kernel and see if you can notice a difference.

dlech, 2016-10-22:
Closing since there have not been any new reports for 1 year. Please reopen if this is still a problem.

closed 2016-10-22 by dlech