ev3dev/ev3dev #410: ev3dev Free Memory

jstougaard, 2015-10-05:
The operating system + services seems to be taking up almost all the available memory on the EV3.

I'm running the newest version of the kernel (ev3-ev3dev-jessie-2015-09-13), with a plugged in wifi adapter.
When running `free` it only reports around 1,5 mb of free memory, when it is idle in the brickman menu (of course running ssh, but nothing else). Is this normal? 
Doesn't seem like a lot to work with, for running my own programs. Or maybe I'm just too used to modern day computers with plenty of spare memory...

Result of `free`:
```
             total       used       free     shared    buffers     cached
Mem:         59464      57856       1608        164       3500      34668
-/+ buffers/cache:      19688      39776
Swap:       139260        856     138404
```

Result of `ps aux` (top items):
```
USER       PID %CPU %MEM    VSZ   RSS TTY      STAT START   TIME COMMAND
root        62  0.4 16.0  20356  9524 ?        Ss   08:19   0:09 /lib/systemd/systemd-journald
root       349  4.0 12.4  55476  7432 tty1     Ssl+ 08:20   1:35 /usr/sbin/brickman
root       528  0.2  8.8  12048  5248 ?        Ss   08:48   0:01 sshd: root@pts/0
root       289  0.1  7.9   6828  4720 ?        Ss   08:20   0:03 /sbin/wpa_supplicant -u -s -O /run/wpa_supplicant
root       459  0.1  7.6  22556  4532 ?        Ss   08:20   0:02 /usr/sbin/nmbd -D
ntp        328  0.0  6.0   5816  3616 ?        Ss   08:20   0:01 /usr/sbin/ntpd -p /var/run/ntpd.pid -g -u 100:103
root       277  0.1  6.0   7840  3572 ?        Ss   08:20   0:03 /usr/sbin/connmand -n
root         1  0.3  5.7  23784  3424 ?        Ss   08:19   0:08 /sbin/init splash
root       548  0.0  5.1   5092  3056 pts/0    Ss   08:48   0:00 -bash
root       539  0.0  4.9   4420  2936 ?        Ss   08:48   0:00 /lib/systemd/systemd --user
root       294  0.0  4.7   7832  2836 ?        Ss   08:20   0:00 /usr/sbin/sshd -D
message+   234  0.2  4.3   5624  2600 ?        Ss   08:20   0:04 /usr/bin/dbus-daemon --system --address=systemd: --nofork --nopidfile --sys
avahi      231  0.2  4.3   3892  2568 ?        Ss   08:20   0:06 avahi-daemon: running [ev3dev.local]
root       230  0.0  3.9   3740  2332 ?        Ss   08:20   0:00 /lib/systemd/systemd-logind
root       279  0.0  3.6   4728  2156 ?        Ss   08:20   0:00 /usr/lib/bluetooth/bluetoothd
root       562  0.0  3.6   4600  2156 pts/0    R+   08:59   0:00 ps aux --sort -rss
root       544  0.0  3.0  25144  1832 ?        S    08:48   0:00 (sd-pam)
root        82  0.0  2.9  10612  1776 ?        Ss   08:19   0:01 /lib/systemd/systemd-udevd
avahi      246  0.0  1.6   3764  1004 ?        S    08:20   0:00 avahi-daemon: chroot helper
root       265  0.0  0.9   1932   576 ?        S<   08:20   0:00 /usr/bin/hciattach -t 30 /dev/ttyS2 texas 2000000 noflow nosleep
```

dlech, 2015-10-05:
It was not too long ago I thought the same thing. In fact since this is so not-obvious, someone devoted an entire website to it: <http://www.linuxatemyram.com/>.

jstougaard, 2015-10-05:
Good to know. Thank you!

closed 2015-10-05 by jstougaard