ev3dev/ev3dev #412: Need a way to detect BrickPi and PiStorms on Raspberry Pi

dlech, 2015-10-19:
This came up in a conversation with @ddemidov in <https://github.com/ev3dev/ev3dev-lang/commit/ac9a629ab3cd8f3681e790313571f655316921ad#commitcomment-13858910>. We want to be able to use the same image file for both BrickPi and PiStorms. Users will have to select which one is attached by changing a setting in [config.txt](https://github.com/ev3dev/brickstrap/blob/master/rpi-ev3dev-jessie/root/boot/flash/config.txt). PiStorms will be easy to detect since is uses a I2C/NXT connection. Basically, it will be automatically detected and everything should "just work". The BrickPi on the other hand uses a tty line discipline. This is attached by a systemd service. So, we need something that can trigger the systemd service rather than the user having to manually enable or disable it. Right now, the presence of the brickpi LED devices is the only thing we have to go off of. It seems like there should be a better way to do this, but I haven't come up with anything yet. Just throwing this out there in case there are any device tree experts listening that may have a better idea.

dlech, 2015-11-04:
OK. Recommended way to know if you are running on...

EV3: The file `/proc/cpuinfo` contains the string `LEGO MINDSTORMS EV3 Programmable Brick`.

    cat /proc/cpuinfo | grep -q "LEGO MINDSTORMS EV3 Programmable Brick"

BrickPi: The file `/sys/class/tty/ttyAMA0/device/of_node/brickpi,status` exists and contains `okay`. (Yes, that is a comma and it is part of the file name.)

    [ -f /sys/class/tty/ttyAMA0/device/of_node/brickpi,status ] && [ $(cat /sys/class/tty/ttyAMA0/device/of_node/brickpi,status) == "okay" ]

PiStorms: The directory `/sys/bus/i2c/drivers/pistorms` exists and contains the symlink (directory) `?-001a` (and  `?-001b`).

    [ -d /sys/bus/i2c/drivers/pistorms ] && [ -L /sys/bus/i2c/drivers/pistorms/?-001a ]

ddemidov, 2015-11-04:
Can we introduce a system shell script (`ev3dev_platform`?) that would return `ev3`/`brickpi`/`pistorms` based on this? The script could be universally used by all the bindings.

rhempel, 2015-11-04:
Can we introduce a more general form of this platform detection? I know that in some cases, the DKMS drivers are all that a user will have, but for the rest of the platforms, we actually have control of the image on the SD card, and therefore we control the kernel.

Can we not just add a known string to `/proc/cpuinfo` or create our own /proc/ev3dev` or `/sys/class/ev3dev/platform` that has the version string?

dlech, 2015-11-04:
> Can we introduce a system shell script (ev3dev_platform?) that would return ev3/brickpi/pistorms based on this? The script could be universally used by all the bindings.

Yes, we can do this.

> Can we introduce a more general form of this platform detection?

No, we can't. If users are using DKMS drivers, then they just don't have a specific platform. You will never get EV3 LEDs or BrickPi LEDs or PiStorms LEDs using DKMS. These are defined by the board configuration file in the kernel in the case of EV3 and in device-tree overlays in the case of Raspberry Pi. And in fact, your question has made me realize that my proposed detection for the PiStorms is really too general as it is and needs to be revised.

> Can we not just add a known string to /proc/cpuinfo

This is exactly what has been done in the case of the EV3. But on Raspberry Pi, this can only tell us which model of Raspberry Pi we have. It cannot tell if we have a BrickPi or a PiStorms or neither.

> or create our own /proc/ev3devor/sys/class/ev3dev/platform` that has the version string?

This is exactly what I have done in the case of the BrickPi. The file `/sys/class/tty/ttyAMA0/device/of_node/brickpi,status` only exists when the BrickPi overlay is enabled. I'm thinking I should do the same for PiStorms. However it doesn't use the tty. It uses i2c and spi. The i2c can be one of 2 devices depending on which Pi model you have which was the reason for the `?`, but perhaps we can attach a custom attribute to the spi instead.

dlech, 2015-11-04:
> Can we introduce a system shell script (ev3dev_platform?) that would return ev3/brickpi/pistorms based on this? The script could be universally used by all the bindings.

However, as @rhempel mentioned, people can be using ev3dev on non-ev3dev platforms, so depending on an external tool is probably not the best idea.

rhempel, 2015-11-04:
Does each supported platform need a custom kernel?

dlech, 2015-11-04:
In the case of the EV3, obviously yes. In the case of Raspberry Pi, no. However, on the Raspberry Pi in order to have the full "platform" for BrickPi or PiStorms, you have to have the device-tree overlays (which is sort of like a custom kernel).

rhempel, 2015-11-04:
Sooo - can we put a virtual "thing" in the device tree that forces a kernel module to be required, and all that kernel module does is create a node somewhere in `/sys` that lets us identify the hardware?

dlech, 2015-11-04:
I guess what you are getting at is we should have one single place to look for all platforms? That is probably not be possible with device tree without a custom kernel.

dlech, 2015-11-05:
Well, after thinking about it some more, of course we could do that because we would have the driver in our DKMS package. So, the only problem I see with that is what happens when someone puts their PiStorms on their BrickPi and both overlays try to create the same device node.

rhempel, 2015-11-05:
Ideally one single place to look for a platform identifier, yes. I see your point re rPi - depending on which shield you have you get a different platform identifier.

If we're worried about someone putting BOTH the PiStorms and the BrickPi on the rPi at once we can do two things. First, we say "Don't do that - it's unsupported!" Or we can make a `/sys/class/ev3dev/` node and allow each overlay to create its own attribute:

- `/sys/class/ev3dev/ev3-mono`
- `/sys/class/ev3dev/ev3-pistorms`
- `/sys/class/ev3dev/ev3-brickpi`

etc?

dlech, 2015-11-05:
I'll have to think about the mechanics of it some more, but I am on board with doing something like this.

rhempel, 2015-11-05:
And if you've got DKMS drivers on your PC - `/sys/class/ev3dev/generic`

dlech, 2015-11-06:
Before I spend time implementing this, I want to make sure we are actually solving a problem that needs solving. I've already solved my problem for bringing this up, which was how do I keep `brickpi.service` from running when using PiStorms.

Here is a comparison of "platforms" so you can see the big picture.

| platform | EV3 | BrickPi(+) | PiStorms | EVB |
|---|---|---|---|---|
| cpu board | N/A (integrated) | Raspberry Pi (all models) | Raspberry Pi (all models) | Beaglebone Black (possibly "white" as well?) |
| display | 178x128 1bpp 1==black (160x128 16bpp RGB565 with hack) | N/A (but possible to add another board with display) | 320x240 16bpp RGB565 | 220x176 16bpp RGB565  |
| fbdev | `/dev/fb0` | N/A | `/dev/fb1` | `/dev/fb0` |
| buttons | 6 | 0 | 1 (plus touchscreen) | 6 |
| backspace | yes | no | no | yes |
| enter | yes | no | yes | yes |
| up/down/left/right | yes | no | no | yes |
| touch | no | no | yes | no |
| sound | yes | no  | no | yes |
| cpu board sound | N/A | line out or HDMI | line out or HDMI | HDMI only |
| input ports | 4 | 4 (plus one I2C-only port on BrickPi) | 4 (not labeled 1,2,3,4 like other platforms - BAS1,BAS2,BBS1,BBS2 instead) | 4 |
| output ports | 4 | 4 | 4 (not labeled A,B,C,D like other platforms - BAM1,BAM2,BBM1,BBM2 instead) | 4 |
| leds | 2 - red/green (4 individual) | 2 - blue | 2 - red/green/blue (6 individual) | 0 |
| cpu board leds | N/A | 1 - red (plus 1 - green on some RPi models) | 1 - red (plus 1 - green on some RPi models) | 4 - blue |
| power supply | `legoev3-battery` | `brickpi-battery` (only on BrickPi+) | `pistorms-battery` | `evb-battery` |
| voltage | yes | yes | yes | yes |
| current | yes | no | no | yes? |

dlech, 2015-11-06:
Here is how I would go about detecting the major components.

Display: Don't care what the "platform" is, just read fbinfo via ioctl. Different fbdev can be handled via `FRAMEBUFFER` environment variable or other OS means as it is in other programs.

Buttons: `/sys/class/input/input*/name`. It will tell you the "platform".

Sound: Don't care what the "platform" is. The default sound card will be configurable via brickman or alsa command line utils.

LEDs: Look at `/sys/class/leds`. Ignore any leds that do not end with `:ev3dev`. The first part of the name will tell you what the platform is.

Battery: Don't care what the "platform" is. Look for power supply with `POWER_SUPPLY_PROP_SCOPE == POWER_SUPPLY_SCOPE_SYSTEM`. That is the "system" power supply.

In/Out ports: Other than the different port names on PiStorms, these are basically the same. Not sure the best way to handle this yet.

dlech, 2015-11-06:
@ensonic, how do you see this fitting in to OpenRoberta?

ddemidov, 2015-11-06:
I like this. I think we won't loose any functionality this way except for led grouping. I guess there is no generic way to tell that some leds form a group that may be combined to control color? Of course we could put all possible groupings into specification, and only use those that do exist on the current platform.

dlech, 2015-11-06:
Actually, you can tell led groupings. The standard format for led names is `devicename:color:function` (comes from https://www.kernel.org/doc/Documentation/leds/leds-class.txt).

`devicename` can contain `:` characters, so when parsing you need to search for the last two `:`s. For leds that can be controlled by ev3dev, the `function` will be `ev3dev`. All other leds should be ignored. Led groups will have the same `devicename` but a different `color`.

ddemidov, 2015-11-06:
These are leds names I know about:

**EV3**:
```
ev3-left0:red:ev3dev
ev3-left1:green:ev3dev
ev3-right0:red:ev3dev
ev3-right1:green:ev3dev
```
**BrickPi**:
```
brickpi1:blue:ev3dev
brickpi2:blue:ev3dev
```
**PiStorms**:
```
pistorms:BA:red:ev3dev
pistorms:BA:blue:ev3dev
pistorms:BA:green:ev3dev
pistorms:BB:red:ev3dev
pistorms:BB:green:ev3dev
pistorms:BB:blue:ev3dev 
```

For BrickPi and PiStorms the group names can be easily derived from led names (`brickpi1`/`brickpi2` and `pistroms:BA`/`pistorms:BB`). For ev3 its not so obvious: each led gets unique device name. We could strip device names of trailing digits (and get `ev3-left`/`ev3-right` groups for ev3), but that would also break grouping for brickpi.

Is it possible to rename leds on ev3 to be
```
ev3-left:red:ev3dev
ev3-left:green:ev3dev
ev3-right:red:ev3dev
ev3-right:green:ev3dev
```
?

ensonic, 2015-11-06:
RIght now the blockly blocks on roberta are quite statically modeled to work for the ev3. We can probably do things to have more dynamic led blocks easilly.

@dlech on ev3dev, could we drop the numbers? So that we have ev3-{left,right}:{red,green}:ev3dev. To an app that reads like left and right led with r/g controls. The facts that it is two separate leds instead one combined is an impl detail.

rhempel, 2015-11-06:
I agree with @ddemidov and @ensonic - the LED number is an implementation detail.

Now if the group of LEDs on the left side of the board were arranged in a line (think bars of reception or battery indicator) then an index number would be reasonable.

Let's think about how to specify LED groups in `spec.json` in a general way and use the spec to generate platform specific code.

Actually, even if we don't autogenerate the code, thinking about the LED groups in the spec will probably help us come up with a better naming scheme.

dlech, 2015-11-06:
Yes. I'm thinking `ev3:{left,right}:{red,green}:ev3dev` (colon after the `ev3` instead of dash) to make it more like the pistorms.

And the more I'm thinking about this, detecting each component individually makes more sense than trying to define an entire platform from a single property. Not only are there nuances between each platform, there are already nuances within each platform (e.g. color screen on EV3 and BrickPi/BrickPi+). And BrickPi and PiStorms have frimware that can be upgraded, so it is possible there could be even more differences and we would have to look at firmware version too. I gets very complex very quickly. I think just looking at each subsystem individually is going to work better in the long run.

ddemidov, 2015-11-06:
> Buttons: /sys/class/input/input*/name. It will tell you the "platform

It says `EV3 buttons` and `EV3 speaker beep` on EV3. Is there a way to know actual buttons supported on the system, or should we keep a database of platform keys?

> In/Out ports.

If I do `cat /sys/class/lego-port/*/port_name` on EV3, I get
```
in1
in2
in3
in4
outA
outB
outC
outD
```
Is lego-port class supported on all platforms, or just ev3?

dlech, 2015-11-06:
> Is there a way to know actual buttons supported on the system

Yes, `/sys/class/input/input0/capabilities/key` returns a bitmask of the keys that are present. the `EVIOCGBIT(ev,len)` ioctl does the same thing.

> or should we keep a database of platform keys?

If you want the keys available via introspection while developing on a host computer, then probably yes.

> Is lego-port class supported on all platforms, or just ev3?

The lego-port class is on on all platforms. Additionally, if you have any kind of sensor multiplexer, you will have additional ports, like `in1:i2c8:mux2`, etc. On BrickPi (for now) the port names are the same (`in1', `outA`), but on PiStorms, they are `pistorms:BAS1`, `pistorms:BAM1`.

dlech, 2015-11-09:
Fix released in kernel cycle 8 (`*-8-ev3dev-*`).

closed 2015-11-09 by dlech

EricPobot, 2015-12-04:
For your information, I've just received a brand new PiStorms, and the RGB LED on the left side (bank B) is not here any more on the new version of the board (black PCB). 
I firstly suspected my board to be damaged, since not being able to activate it using the provided Python lib, but the support team answered me with this information after asking them.

dlech, 2015-12-04:
please open a new issue for this.

EricPobot, 2015-12-04:
I was just giving this as an info, since this thread seemed to talk about board identification. It was not an issue in my mind.