ev3dev/ev3dev #414: Balancing robot

dpiquee, 2015-10-22:


dpiquee, 2015-10-22:
Hi there
I have experimented with great pleasure the fabulous laurens walk balancing robot
http://robotsquare.com/2014/06/23/tutorial-building-balanc3r/
I would like work in python and ev3dev,
I was inspired to http://www.ev3dev.org/projects/2014/05/16/Balancing-Robot/
but without success (motor attributs does not match...)
Has anyone succeeded?
a java solution also interest me
Thanks for your attention

dlech, 2015-10-22:
Looks like my code is a bit out of date at <https://gist.github.com/dlech/11098915>. I've just updated it, but I have not tested it, so there could still be problems. You can find updated documentation on how to use the motors at <http://www.ev3dev.org/docs/drivers/tacho-motor-class/> and <http://www.ev3dev.org/docs/tutorials/tacho-motors/>.

You could also try out the new ev3dev-lang python library. See ongoing discussion at <https://github.com/ev3dev/ev3dev-lang/issues/104>.

For java, your best bet is to use [leJOS](http://www.lejos.org). I don't know of anyone working on java with ev3dev.

dpiquee, 2015-10-23:
INIT method of the class EV3 MOTOR does not work for me ...
I have tinkered one
and it is much better
the engines now meet
the robot is a little crazy, but he moves ...
Thanks for your help

closed 2015-10-23 by dpiquee