ev3dev/ev3dev #416: ev3dev as firmware for the EV3 - aka ev3devLite

dlech, 2015-10-23:
@botbench (product manager of ROBOTC) has expressed interest in using ev3dev as the OS for ROBOTC, but only if it can be installed as firmware rather that being installed on a microSD card. The primary concern is that microSD cards in a classroom full of kids is asking for problems. If you have ever tried teaching robotics to children, you won't have any trouble imaging the things they will do. And, now that OpenRoberta seems to be moving to ev3dev as well, I think this could be a benefit to that project as well.

To make this work, we basically need a kernel configuration and root file system that are trimmed down to fit in the 16MB flash and still leave room for some user programs. I think we can still use brickstrap and debian packages to make the root file system, although we probably want to take a hint from the now defunct emdebian repositories and strip out all of the docs.

Here are the tasks I have identified that are needed to make this happen:

- [ ] Create a new `ev3dev_lite_defconfig` that contains the kernel configuration.
- [ ] Create a new board definition in brickstrap.
- [ ] Copy the uboot code from lms2012
- [ ] Make a script that takes the uboot code and the tar file produced by brickstrap and turns it into a firmware image.

rhempel, 2015-10-24:
@dlech,  do we want to implement this as a branch of the `ev3dev-kernel` - or do we want to make a whole new repo as if it was a different platform?

dlech, 2015-10-24:
No need for that. We just need a separate kernel configuration file. It can be built from the same source code. We're actually doing exactly that now with RPi1 and RPi2.

rhempel, 2015-10-24:
OK, good - so a monolithic kernel? Updates will require reflashing anyways, and we can strip out a ton of extras that they won't need.

dlech, 2015-10-25:
I just came across this: http://buildroot.uclibc.org/

I think it may be the way to go. We just have to hook it up so that it uses our kernel and uboot from LEGO. That should be enough to get a bootable system, then we can add custom ev3dev packages as needed, like brickstrap/ev3devKit/GRX.

rhempel, 2015-10-26:
Does that buildroot replace any of our other tools for building the big `ev3dev` image?

dlech, 2015-10-26:
Yes, buildroot will build *everything*, so will be used instead of the existing tools (for ev3devLite only).

dlech, 2016-02-18:
Look ma, no SD card! Here is an ev3dev image that you can install in the flash memory of your EV3. Unzip the file and use the official LEGO MINDSTORMS EV3 software to load it.

[ev3dev-lite.zip](https://github.com/ev3dev/ev3dev/files/135709/ev3dev-lite.zip)

There are some serious issues with this, like no brickman, I2C sensors cause a lock up, sound doesn't work, no bluetooth, UART on ports 3,4 doesn't work and so on. But if you know what you are doing, you can get EV3 sensors and motors to work. Default root password is `r00tme`.

To re-flash the EV3 after installing this image, hold down the back, enter and right buttons on the EV3 until it turns off, then release the enter button first, then the other buttons.

Anyone interested in pursuing this further?

botbench, 2016-02-18:
This is all kinds of awesome! 

What are the reasons for these issues?

rhempel, 2016-02-18:
Wow! Nice work David! I figured Xander would be interested in pursuing this further...

dlech, 2016-02-18:
Most of the issues are actually only serious from the point of view of usability but not from the point of view of fixing them. It is mostly that there are not buildroot packages for custom EV3 stuff. We need the firmware files for Bluetooth and soft UARTs, and some init scripts and udev rules. This will fix most things. Then we need to package grx, ev3devKit and brickman. These could be a bit tricky since they actually have to be compiled and the latter two are written vala (although I have recently fixed those to make cross compiling easier).

The only really serious issue is the I2C lockup. PCM playback also locks up, so this tells me that it has to do with using FIQ. However I would actually rather pursue using the PRU instead of FIQ rather than trying to figure out what is wrong with FIQ (see #506).

rhempel, 2016-02-18:
For Xander's purposes, they already have a UI if I understand it correctly - the missing link is a modern kernel and the file system interface to devices that the ev3dev kernel (and its modules) provides.

Is that more or less right, Xander?

dlech, 2016-02-18:
It turns out fixing the FIQ lockup was easy (was loading the kernel image to the wrong address). Here is a new image for testing. [ev3dev-lite.zip](https://github.com/ev3dev/ev3dev/files/136838/ev3dev-lite.zip)

So, I really don't see any blocker for moving forward with this if someone is willing to put in the time.

botbench, 2016-02-19:
It turns out fixing the FIQ lockup was easy (was loading the kernel image to the wrong address). Here is a new image for testing. ev3dev-lite.zip <https://github.com/ev3dev/ev3dev/files/136838/ev3dev-lite.zip>

That’s great news! Thanks so much!
For Xander's purposes, they already have a UI if I understand it correctly - the missing link is a modern kernel and the file system interface to devices that the ev3dev kernel (and its modules) provides.

Is that more or less right, Xander?

We don’t have a UI, as we use the LEGO’s VM to do that, but that would be the least of our issues.  If we have working I2C, sound and all that fun stuff, that wouldn’t be all that important.  We could make something pretty quickly, I’m sure.  We could modify the one for the NXT, as it’s baked into the ROBOTC NXT VM.  All of the VMs in ROBOTC use a generic interface to the underlying video buffer hardware.

I’ll try to take a look this weekend, but I am on super slow internet right now, so downloading stuff is painful.  I’ll be back in the Netherlands next Saturday or Sunday, though, where I have a 55Mbit/5Mbit VDSL2 line.

Regards,
Xander

dlech, 2016-11-17:
Quick update: I sent some patches to builtroot recently. You can now use the official buildroot software to build a firmware image (and sd card image) that will actually boot on the EV3. Using buildroot >= 2016.11-rc1...

    make lego_ev3_defconfig 
    make

It will take 30 minutes to several hours to build depending on how fast your machine is. This will produce a firmware file `output/images/flash.bin` and an SD card image `output/images/sdcard.img`.

markyi370, 2017-05-16:
Any new updates?

dlech, 2017-05-16:
I just sent a patch to the buildroot mailing list yesterday to use the mainline linux kernel instead of the ev3dev kernel. You can't do much with it, but it's out there.

dlech, 2017-08-12:
Unless someone takes an interest in maintaining it, we probably won't ever release an "ev3devLite". But anyone interested in creating a flash image can do so with Buildroot.

closed 2017-08-12 by dlech