ev3dev/ev3dev #418: Cannot run files owned by system users

dlech, 2015-10-30:
Originally posted by @TomHartogs at <https://github.com/ev3dev/ev3dev-lang/issues/110>.

>I just created and compiled a C++ program writing "Hello World!" in the console. However when I try to launch the same programm from the brick itself I get an error message saying "Cannot run files owned by system users". I recall being able to run a Hello World program on the brick before. I already tried changing the permissions. Is this happening because I am working in the root folder?

dlech, 2015-10-30:
Fixed in https://github.com/ev3dev/brickman/commit/cdf74c04ec8e710e835db22c2445c37a5588fd6c and https://github.com/ev3dev/brickman/commit/c5ad45e64b134c358fa25be1a3889b5497d595ba

dlech, 2015-10-30:
Before these changes, brickman would run programs as the user that was the owner of the file. Starting with the next release, all program will be run with the `robot` user regardless of who owns the file.

ddemidov, 2015-10-30:
Would  robot be a privileged user, or we would have to explicitly give the user access to the files we want to run?

dlech, 2015-10-30:
The `robot` user will be installed by default. There will no longer be a default root password, so users will have to log in as `robot`. `robot` will be a member of [lots of groups](https://github.com/ev3dev/ev3dev-adduser-config/blob/ev3dev-jessie/adduser.conf.ev3dev#L77), so should have no problems running most programs.

Brickman will no only display files in `/home/robot` instead of all of `/home`. So, to be able to run a file, it just needs to be under `/home/robot`. Of course, this could just be a symlink to a program somewhere else.

dlech, 2015-12-16:
Fix has been released in brickman 0.7.0. More casual users will want to wait for the next image release before updating. See the [announcement](http://www.ev3dev.org/news/2015/12/15/Package-Release/) for more info.

closed 2015-12-16 by dlech